android/camera-samples #77: Camera Preview is Pixeleted

aqsashahid123, 2019-10-09:
In some devices like nokia 1.6 and samsung galaxy A10 the preview resolution is very low.
However it is working fine in all other devices.
The resolution is very very low.

owahltinez, 2019-10-09:
This sounds like https://issuetracker.google.com/issues/140194349 which has now been fixed but likely won't be included in the library until `alpha06`

waliahimanshu, 2019-10-18:
Tested on Samsung device and this is fixed in `alpha06`. The api has changed  and sample app still used `alpha05`.

owahltinez, 2019-10-29:
Thanks for testing that @waliahimanshu! We are working on updating the sample, and it will use `alpha06` soon.

owahltinez, 2019-11-21:
We have a PR under review which will fix this issue: #97

owahltinez, 2019-11-27:
The sample has been updated to `alpha06`, please reopen if this is still an issue

closed 2019-11-27 by owahltinez

askarsyzdykov, 2020-01-20:
Hi @owahltinez! I have similar issue with `alpha06` and `alpha08`

This is preinstalled Android 9 app
[image]

This is sample:
[image]

As you can see, the quality on the second picture a bit pixelated.