android/camera-samples #78: [CameraX] Can't open gallery

k0ur0x, 2019-10-10:
sometimes when I try to open the gallery, the app closes unexpectedly. this often happens after changing device orientation.

Steps to reproduce:
1. change device orientation
2. try to open the gallery
(you may need to repeat these steps to reproduce the bug)

Stacktrace:
`2019-10-10 13:00:00.971 22386-22386/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.cameraxbasic, PID: 22386
    java.lang.IllegalArgumentException: navigation destination com.android.example.cameraxbasic:id/action_camera_to_gallery is unknown to this NavController
        at androidx.navigation.NavController.navigate(NavController.java:789)
        at androidx.navigation.NavController.navigate(NavController.java:730)
        at androidx.navigation.NavController.navigate(NavController.java:716)
        at androidx.navigation.NavController.navigate(NavController.java:907)
        at com.android.example.cameraxbasic.fragments.CameraFragment$updateCameraUi$4.onClick(CameraFragment.kt:381)
        at android.view.View.performClick(View.java:6663)
        at android.view.View.performClickInternal(View.java:6635)
        at android.view.View.access$3100(View.java:794)
        at android.view.View$PerformClick.run(View.java:26199)
        at android.os.Handler.handleCallback(Handler.java:907)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:216)
        at android.app.ActivityThread.main(ActivityThread.java:7625)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)`

owahltinez, 2019-10-16:
Thanks for reporting this! I can confirm that I can reproduce the issue. It appears to be an issue with the navigation library as reported here: https://stackoverflow.com/questions/51060762/java-lang-illegalargumentexception-navigation-destination-xxx-is-unknown-to-thi

closed 2019-12-05 by owahltinez

mickod, 2019-12-27:
Has this issue been fixed - the link the to problem on Stack Overflow highlights problems with fragment navigation and some workarounds but its not clear if any have been applied here or if the issue in this project has been solved some other way?

commit d5247d0a (2019-12-05): Fix nullable calls and back navigation

- Fixes #78
- Fixes CI build failures

Change-Id: I28c4bac48c9e616c802dde19b695edc7eef5884b