android/camera-samples #81: [CameraX] android.os.ServiceSpecificException: supportsCameraApi:1741: Unknown camera ID 1 (code 3)

JonathanImperato, 2019-10-14:
Please follow the following instructions before filing a bug:

When I start any Camera Sample on my device the app crashes.
Here the stack trace:

> 
    java.lang.RuntimeException: Unable to get provider androidx.camera.camera2.impl.Camera2Initializer: java.lang.IllegalArgumentException: supportsCameraApi:1741: Unknown camera ID 1
        at android.app.ActivityThread.installProvider(ActivityThread.java:6283)
        at android.app.ActivityThread.installContentProviders(ActivityThread.java:5846)
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5763)
        at android.app.ActivityThread.-wrap1(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1690)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6548)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:857)
     Caused by: java.lang.IllegalArgumentException: supportsCameraApi:1741: Unknown camera ID 1
        at android.hardware.camera2.CameraManager.throwAsPublicException(CameraManager.java:650)
        at android.hardware.camera2.CameraManager.getCameraCharacteristics(CameraManager.java:245)
        at androidx.camera.camera2.impl.SupportedSurfaceCombination.generateSupportedCombinationList(SupportedSurfaceCombination.java:941)
        at androidx.camera.camera2.impl.SupportedSurfaceCombination.init(SupportedSurfaceCombination.java:603)
        at androidx.camera.camera2.impl.SupportedSurfaceCombination.<init>(SupportedSurfaceCombination.java:88)
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.init(Camera2DeviceSurfaceManager.java:192)
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.<init>(Camera2DeviceSurfaceManager.java:64)
        at androidx.camera.camera2.Camera2AppConfig.create(Camera2AppConfig.java:58)
        at androidx.camera.camera2.impl.Camera2Initializer.onCreate(Camera2Initializer.java:45)
        at android.content.ContentProvider.attachInfo(ContentProvider.java:1919)
        at android.content.ContentProvider.attachInfo(ContentProvider.java:1894)
        at android.app.ActivityThread.installProvider(ActivityThread.java:6280)
        at android.app.ActivityThread.installContentProviders(ActivityThread.java:5846) 
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5763) 
        at android.app.ActivityThread.-wrap1(Unknown Source:0) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1690) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.app.ActivityThread.main(ActivityThread.java:6548) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:857) 
     Caused by: android.os.ServiceSpecificException: supportsCameraApi:1741: Unknown camera ID 1 (code 3)
        at android.os.Parcel.readException(Parcel.java:2027)
        at android.os.Parcel.readException(Parcel.java:1959)
        at android.hardware.ICameraService$Stub$Proxy.supportsCameraApi(ICameraService.java:549)
        at android.hardware.camera2.CameraManager.supportsCameraApiLocked(CameraManager.java:703)
        at android.hardware.camera2.CameraManager.supportsCamera2ApiLocked(CameraManager.java:679)
        at android.hardware.camera2.CameraManager.getCameraCharacteristics(CameraManager.java:228)
        at androidx.camera.camera2.impl.SupportedSurfaceCombination.generateSupportedCombinationList(SupportedSurfaceCombination.java:941) 
        at androidx.camera.camera2.impl.SupportedSurfaceCombination.init(SupportedSurfaceCombination.java:603) 
        at androidx.camera.camera2.impl.SupportedSurfaceCombination.<init>(SupportedSurfaceCombination.java:88) 
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.init(Camera2DeviceSurfaceManager.java:192) 
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.<init>(Camera2DeviceSurfaceManager.java:64) 
        at androidx.camera.camera2.Camera2AppConfig.create(Camera2AppConfig.java:58) 
        at androidx.camera.camera2.impl.Camera2Initializer.onCreate(Camera2Initializer.java:45) 
        at android.content.ContentProvider.attachInfo(ContentProvider.java:1919) 
        at android.content.ContentProvider.attachInfo(ContentProvider.java:1894) 
        at android.app.ActivityThread.installProvider(ActivityThread.java:6280) 
        at android.app.ActivityThread.installContentProviders(ActivityThread.java:5846) 
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5763) 
        at android.app.ActivityThread.-wrap1(Unknown Source:0) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1690) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.app.ActivityThread.main(ActivityThread.java:6548) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:857) 

Device:
Umidigi Z2, Android 8.1.0

JonathanImperato, 2019-10-15:
CameraX Dev group issue link: https://groups.google.com/a/android.com/forum/?utm_medium=email&utm_source=footer#!topic/camerax-developers/O3H-doE_G8Y

owahltinez, 2019-10-29:
Thank you @JonathanImperato, closing this issue since it's unrelated to the sample (potential culprit is wither the library or the framework)

closed 2019-10-29 by owahltinez