android/camera-samples #82: Camera2Basic not working on emulator?

Monism1, 2019-10-17:
I'm trying Camera2Basic Java on the emulated Nexus S, API  27, Android 8.1 on Android studios, the app loads fine but when clicking picture the image does not save (I don't receive any toast message saying "saved" or "error" or anything), there does not appear to be any error posted in the logs. (I have tried the same on a few other emulated devices and same issue). Thanks in advance for any help.

owahltinez, 2019-10-21:
Thanks for reporting this! I can confirm that I'm able to reproduce the issue you described. Strangely, it only occurs in the emulator. While we look into it, you can use the Camera2Formats sample as a reference for a simple photo-taking app -- although that one is Kotlin-only.

owahltinez, 2020-01-24:
We have updated Camera2Basic, now it works on emulator

closed 2020-01-24 by owahltinez