android/camera-samples #83: setTargetAspectRatio(Rational) don't work in latest version

anfilin, 2019-10-17:
Latest CameraX version (1.0.0-alpha06) require AspectRation object (not Rational) for setTargetAspectRatio method. Could you fix it?

pjahnmodum, 2019-10-17:
Using setTargetAspectRatioCustom worked for me

anfilin, 2019-10-17:
> Using setTargetAspectRatioCustom worked for me

For 1.0.0-alpha06 or for previous version? New version requires AspectRation object for setTargetAspectRatio.

[](https://developer.android.com/reference/kotlin/androidx/camera/core/ImageAnalysisConfig.Builder#settargetaspectratio
)

pjahnmodum, 2019-10-18:
For alpha06. 
The method setTargetAspectRatio(Rational aspectRatio) has been changed for ImageCapture to setTargetAspectRatioCustom(Rational aspectRatio). When set, the ImageCapture output will be cropped accordingly.
https://developer.android.com/jetpack/androidx/releases/camera#camera2-core-1.0.0-alpha06
Just change it and it should work again

anfilin, 2019-10-18:
Thank you. It works.

closed 2019-10-18 by anfilin