android/camera-samples #88: Delay in photo capture.

ujwal-coditas, 2019-10-23:
When I compared this sample with the stock camera apps of any manufacturer, this sample takes time to capture the image after the click on the button. When I capture an image of object A and after the click, if I move the camera to object B, I am getting an image of object B. 

When I do the same with the stock camera apps, they are much much faster.

What is the reason behind this delay, how should we achieve better performance while capturing images using Camera2 API?

owahltinez, 2019-10-29:
What device are you seeing this in? Camera2Basic waits for focus convergence before taking every picture. You can also see Camera2Formats, which takes pictures without waiting for the focus to converge and should be as fast as any camera app can be.

ujwal-coditas, 2019-11-01:
Got it. Thanks :)

closed 2019-11-01 by ujwal-coditas

fzyzcjy, 2021-09-02:
> What device are you seeing this in? Camera2Basic waits for focus convergence before taking every picture. You can also see Camera2Formats, which takes pictures without waiting for the focus to converge and should be as fast as any camera app can be.

Hi I wonder how can I wait focus convergence for camerax? thanks!

related https://stackoverflow.com/questions/69029799/how-to-ensure-focused-before-taking-picture-in-android-camerax-should-be-a-ver