android/camera-samples #89: java.lang.IllegalStateException: Detaching occurs more times than attaching

aqsaShahid-zameen, 2019-10-23:
This is my Logcat
  at androidx.camera.core.DeferrableSurface.notifySurfaceDetached(DeferrableSurface.java:160)
        at androidx.camera.camera2.impl.Camera.notifyDetachFromUseCaseSurfaces(Camera.java:667)
        at androidx.camera.camera2.impl.Camera.removeOnlineUseCase(Camera.java:789)
        at androidx.camera.camera2.impl.Camera$13.run(Camera.java:772)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:65)
Everything was working fine(except on few devices) before updated and then I updated to cameraX alpha-006 and now I am getting this error.
Any idea how this is going to get fixed?
Any help will be appreciated.

aqsaShahid-zameen, 2019-10-23:
I think the bug was on my end as i was extending fragment from Executor and using the same thread over and over again whereas now I have created a new executor for everytime the ImageCapture is called and that has pretty much fixed it.

owahltinez, 2019-10-29:
What sample are you seeing this in? What device produces this error? Have you modified the sample in any way?

aqsaShahid-zameen, 2019-11-14:
I have updated camerax to alpha-006.

owahltinez, 2019-11-18:
If you are modifying the sample, then it's not a sample issue. Please use the Camera component under https://issuetracker.google.com

closed 2019-11-18 by owahltinez