android/camera-samples #92: [CameraX] App crashes api < 21

sawyerhard, 2019-10-28:
- Android version 4.4.4, API 19
- dependencies:
androidx.camera:camera-core:1.0.0-alpha06
androidx.camera:camera-camera2:1.0.0-alpha06

**Problem**
I'm trying to use this library only for API > 21, but i also can't run my application for API < 21 . 
I got crash with this exception:
`E/UncaughtException: java.lang.NoClassDefFoundError: android.util.Size`

I'm trying to use this method, but it's not help: [stackoverflow](https://stackoverflow.com/questions/56374939/android-application-with-camera2-library-crash-on-start-for-sdk19)

Merseyside, 2019-10-28:
The same issue

youlovejohnny, 2019-10-28:
Same, fix it asap please

roffel, 2019-10-28:
I would like to be able to build with minSDK 16 as well. It is fine if I need to fall back to an own camera implementation for Android 4.x devices.

waliahimanshu, 2019-10-28:
Camer2 and CameraX api's minimum supported API level is 21.

owahltinez, 2019-10-29:
As @waliahimanshu CameraX is built on Camera2, and Camera2 only supports API 21+

closed 2019-10-29 by owahltinez

sawyerhard, 2019-10-29:
but why did you do so? why i cant use overrideLibrary tools in manifest?

roffel, 2019-10-29:
@owahltinez But before I could write an if statement stating `if < api 21 use the deprecated camera` so I could still build for Android 4.x. That is not possible anymore.

peterdk, 2019-12-02:
Here another vote, I want to use CameraX where possible, but write the code for legacy Camera myself, with a condition SDK_INT check. Now just adding the implementations makes the app crash. I need to support 4.1+, so I really don't understand why it cannot be used conditionally.
```
implementation 'androidx.camera:camera-core:1.0.0-alpha06'
implementation 'androidx.camera:camera-camera2:1.0.0-alpha06'
```

waliahimanshu, 2019-12-04:
@peterdk The crash is due to fact that the cameraX using `size` class called from  the `Camera2Initializer` content provider which is added in api 21.

One way to get around that to create a empty content provider with the same name `Camera2Initializer`
`class Camera2Initializer : ContentProvider() {...}`  

Init the library before starting the camera 
`CameraX.init(applicationContext, Camera2AppConfig.create(applicationContext))`

Having said that, the init process has changes significantly in alpha-07.

You can use this workaround, until you decide to bump the min-sdk version.