android/camera-samples #94: CameraX memory leak

TooLazyy, 2019-10-29:
Download CameraX sample. 
Add new empty activity.
Add LeakCanary dependency and
`LeakSentry.refWatcher.watch(this)` to `onDestroyView` of `CameraFragment`.
 After app started and all permissions granted start new activity and call finish on `MainActivity`.
Like this f.x.
```
        Handler().postDelayed({
            startActivity(Intent(this, TestActivity::class.java))
            finish()
        }, 10000L)
```
You will see CameraX instance is leaking.
Some screens from my app. In your sample leaks are the same.
[image]
[image]
[image]

owahltinez, 2019-11-05:
Thanks for reporting this @TooLazyy, I think these memory leaks are happening at the library layer and not the sample so there's not much we can't do to fix it here :(

Can you open a bug in the [AOSP issue tracker](https://issuetracker.google.com/issues?q=componentid:618491) for the camera library?

owahltinez, 2019-11-21:
Closing since this is a library issue that should be tracked in the corresponding issue tracker.

closed 2019-11-21 by owahltinez

ocarey, 2020-02-06:
Also having issues with CameraX memory leak w/ version alpha06 and alpha09. I was using alpha06+TextureView and upgraded to alpha09+PreviewView today to try and fix the memory leak. Unfortunately, there's still a memory leak related to PreviewView and its underlying TextureView. This is seemingly the exact same leak I had with alpha06 + TextureView. This occurs in my own app & the CameraXBasic example app.

Also adding this to issue: https://issuetracker.google.com/141188637

LeakCanary CameraXBasic:
[image]

ocarey, 2020-02-06:
By adding this to Activity onDestroy() I was able to force a Garbage Collection of my Camera Activity. Version alpha09 w/ CameraXBasic example app:

```
override fun onDestroy() {
        super.onDestroy()
        tearDownCamera()
    }
```

```
private fun tearDownCamera() {
        val cameraProviderFuture = ProcessCameraProvider.getInstance(this)
        cameraProviderFuture.addListener(Runnable {
            val cameraProvider: ProcessCameraProvider = cameraProviderFuture.get()
            cameraProvider.shutdown()
        }, ContextCompat.getMainExecutor(this))
    }
```

Best,
Owen

Colanderr, 2020-03-02:
I am facing the same issue, but @ocarey 's workaround sometimes crashes the app. 
I've also noticed that for every instance of the activity that's leaked, there is also a leaked androidx.lifecycle.ReportFragment, so I'm guessing that it's a bug in the camera lifecycler library (androidx.camera:camera-lifecycle)

Sottti, 2020-05-16:
@ocarey apparently `shutdown` is `@RestrictTo(Scope.TESTS)`