android/camera-samples #96: CameraX sample app crashes when thumbnail hit quickly after taking photo

mickod, 2019-10-30:
Standard App - not modifications.

Google Pixel running Android 10.

To reproduce take photo and hit gallery button quickly afterwards.

Logcat:

019-10-30 17:34:08.658 11671-11671/com.android.example.cameraxbasic D/CameraXBasic: Screen metrics: 1920 x 1080
2019-10-30 17:34:08.729 11671-11671/com.android.example.cameraxbasic D/AutoFitPreviewBuilder: Preview output changed. Size: 1920x1080. Rotation: 0
2019-10-30 17:34:08.743 11671-11708/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [] for camera: 0
2019-10-30 17:34:08.744 11671-11708/com.android.example.cameraxbasic D/Camera: Use cases [Preview:androidx.camera.core.Preview-2a673c89-710a-46ee-bb15-a9a50db90aae, ImageAnalysis:androidx.camera.core.ImageAnalysis-00ac6c1c-3b89-4e89-8356-e9842f3a5727, ImageCapture:androidx.camera.core.ImageCapture-b91d2d71-6f88-45e4-819d-955cae4eaf04] ONLINE for camera 0
2019-10-30 17:34:08.744 11671-11708/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [] for camera: 0
2019-10-30 17:34:08.744 11671-11708/com.android.example.cameraxbasic D/Camera: Resetting Capture Session
2019-10-30 17:34:08.744 11671-11708/com.android.example.cameraxbasic D/Camera: Opening camera: 0
2019-10-30 17:34:08.745 11671-11708/com.android.example.cameraxbasic D/UseCaseAttachState: All use case: [androidx.camera.core.Preview-2a673c89-710a-46ee-bb15-a9a50db90aae, androidx.camera.core.ImageAnalysis-00ac6c1c-3b89-4e89-8356-e9842f3a5727, androidx.camera.core.ImageCapture-b91d2d71-6f88-45e4-819d-955cae4eaf04] for camera: 0
2019-10-30 17:34:08.758 11671-11671/com.android.example.cameraxbasic D/AutoFitPreviewBuilder: View finder layout changed. Size: 1920x1080
2019-10-30 17:34:08.760 11671-11671/com.android.example.cameraxbasic D/AutoFitPreviewBuilder: Applying output transformation.
    View finder size: 1920x1080.
    Preview output size: 1920x1080
    View finder rotation: 90
    Preview output rotation: 0
2019-10-30 17:34:08.799 11671-11708/com.android.example.cameraxbasic D/Camera: Use case Preview:androidx.camera.core.Preview-2a673c89-710a-46ee-bb15-a9a50db90aae ACTIVE for camera 0
2019-10-30 17:34:08.800 11671-11708/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [androidx.camera.core.Preview-2a673c89-710a-46ee-bb15-a9a50db90aae] for camera: 0
2019-10-30 17:34:08.801 11671-11708/com.android.example.cameraxbasic D/Camera: Use case ImageAnalysis:androidx.camera.core.ImageAnalysis-00ac6c1c-3b89-4e89-8356-e9842f3a5727 ACTIVE for camera 0
2019-10-30 17:34:08.801 11671-11708/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [androidx.camera.core.Preview-2a673c89-710a-46ee-bb15-a9a50db90aae, androidx.camera.core.ImageAnalysis-00ac6c1c-3b89-4e89-8356-e9842f3a5727] for camera: 0
2019-10-30 17:34:08.802 11671-11708/com.android.example.cameraxbasic D/Camera: Use case ImageCapture:androidx.camera.core.ImageCapture-b91d2d71-6f88-45e4-819d-955cae4eaf04 ACTIVE for camera 0
2019-10-30 17:34:08.802 11671-11708/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [androidx.camera.core.Preview-2a673c89-710a-46ee-bb15-a9a50db90aae, androidx.camera.core.ImageAnalysis-00ac6c1c-3b89-4e89-8356-e9842f3a5727, androidx.camera.core.ImageCapture-b91d2d71-6f88-45e4-819d-955cae4eaf04] for camera: 0
2019-10-30 17:34:08.803 11671-11708/com.android.example.cameraxbasic D/Camera: CameraDevice.onOpened(): 0
2019-10-30 17:34:08.804 11671-11708/com.android.example.cameraxbasic D/UseCaseAttachState: All use case: [androidx.camera.core.Preview-2a673c89-710a-46ee-bb15-a9a50db90aae, androidx.camera.core.ImageAnalysis-00ac6c1c-3b89-4e89-8356-e9842f3a5727, androidx.camera.core.ImageCapture-b91d2d71-6f88-45e4-819d-955cae4eaf04] for camera: 0
2019-10-30 17:34:08.805 11671-11708/com.android.example.cameraxbasic D/CaptureSession: Opening capture session.
2019-10-30 17:34:08.905 11671-11708/com.android.example.cameraxbasic D/CaptureSession: Attempting to send capture request onConfigured
2019-10-30 17:34:08.905 11671-11708/com.android.example.cameraxbasic D/CaptureSession: Issuing request for session.
2019-10-30 17:34:08.908 11671-11708/com.android.example.cameraxbasic D/CaptureSession: CameraCaptureSession.onConfigured() mState=OPENED
2019-10-30 17:34:08.908 11671-11708/com.android.example.cameraxbasic D/CaptureSession: CameraCaptureSession.onReady()
2019-10-30 17:34:09.153 11671-11671/com.android.example.cameraxbasic D/CameraXBasic: Average luminosity: 44.28642252604167. Frames per second: Infinity
2019-10-30 17:34:09.680 11671-11708/com.android.example.cameraxbasic D/Camera: issue capture request for camera 0
2019-10-30 17:34:09.680 11671-11708/com.android.example.cameraxbasic D/CaptureSession: Issuing capture request.
2019-10-30 17:34:09.848 11671-11671/com.android.example.cameraxbasic D/AndroidRuntime: Shutting down VM
2019-10-30 17:34:09.850 11671-11671/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.cameraxbasic, PID: 11671
    java.lang.IllegalArgumentException: navigation destination com.android.example.cameraxbasic:id/action_camera_to_gallery is unknown to this NavController
        at androidx.navigation.NavController.navigate(NavController.java:789)
        at androidx.navigation.NavController.navigate(NavController.java:730)
        at androidx.navigation.NavController.navigate(NavController.java:716)
        at androidx.navigation.NavController.navigate(NavController.java:907)
        at com.android.example.cameraxbasic.fragments.CameraFragment$updateCameraUi$4.onClick(CameraFragment.kt:380)
        at android.view.View.performClick(View.java:7140)
        at android.view.View.performClickInternal(View.java:7117)
        at android.view.View.access$3500(View.java:801)
        at android.view.View$PerformClick.run(View.java:27351)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7356)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
2019-10-30 17:34:09.862 11671-11671/com.android.example.cameraxbasic I/Process: Sending signal. PID: 11671 SIG: 9

mickod, 2019-10-30:
Similar to this crash but happens without any rotation - #78

VuManhHung97, 2019-11-01:
I also encountered this situation when using camerax at my application

owahltinez, 2019-11-04:
I think this is a duplicate of #78 please reopen if that's not the case

owahltinez, 2019-11-04:
Thanks for reporting!

closed 2019-11-04 by owahltinez

mickod, 2019-12-31:
It looks like this has now been fixed - if anyone is interested in the underlying issue I think this was the cause: https://stackoverflow.com/a/59544963/334402