android/camera-samples #99: How do I capture depth data in Camera2Formats?

wedouglas, 2019-11-06:
I'm at a loss here. The example says it allows for JPEG, RAW, and Depth capture and shows an example of a depth image, but there is seemingly nothing in the code that that sets a camera capture request for depth. I can find nothing in the API documentation about capturing depth either.

The code shows testing for DEPTH_JPEG, but none of the cameras seem to capture that and there is seemingly no way to enable it. Am I missing something here?

I have a Pixel 4.

closed 2019-11-09 by wedouglas

3dmeshviewer, 2019-11-22:
Same for me I have Pixel 4 XL, there is not official description what characteristics Pixel 4 supports [CameraCharacteristics], I bought more expensive phone for development a this new feature is missing or SDk is not valid. Is working sample for somebody?

gf0507033, 2019-11-25:
@wedouglas do you capture depth map?

wedouglas, 2019-11-25:
@gf0507033 No, it's not possible to capture anything other than normal images from what I can tell. There are no actual setting for enabling depth map capture/creation or anything of the sort. There are no characteristics that are depth related.

yanyanjiao2019, 2019-12-01:
I'm using Huawei P30 Pro for Android development and want to get the depth map from the ToF sensor. But it seems like I can't open the ToF camera individually. I want to know what kind of phone will support the sample codes.

NearZXH, 2020-11-18:
In the code, it show testing of `DEPTH_JPEG `, but actually in most of my tested cases, they are `DEPTH16`(in Huawei P30 Pro & P40 Pro). So you can have a look at [this doc](https://developer.android.com/reference/android/graphics/ImageFormat#DEPTH16). And fortunately, i found a good repo to test it, he save my time. [Here is the repo](https://github.com/plluke/tof) and [his post](https://medium.com/swlh/working-with-the-3d-camera-on-the-samsung-s10-5g-4782336783c)