android/camera-samples #102: `setFlashMode` not work

lijialiang, 2019-11-18:
```
def camerax_version = '1.0.0-alpha06'
def camerax_view_version = "1.0.0-alpha03"
implementation "androidx.camera:camera-core:${camerax_version}"
implementation "androidx.camera:camera-camera2:${camerax_version}"
implementation "androidx.camera:camera-view:$camerax_view_version"
```

```
// MainActivity

var cameraFlashMode = FlashMode.OFF

override fun onCreate(savedInstanceState: Bundle?) {
    //...

    findViewById<RelativeLayout>(R.id.switchCameraFlashMode)
            .setOnClickListener {
                when (cameraFlashMode) {
                    FlashMode.OFF -> {
                        cameraFlashMode = FlashMode.ON
                    }
                    FlashMode.ON -> {
                        cameraFlashMode = FlashMode.OFF
                        startCamera()
                    }
        }
    }
}

private fun startCamera() {
        CameraX.unbindAll()

        val previewConfig = PreviewConfig.Builder().apply {
            setLensFacing(cameraLensFacing)
            setTargetResolution(Size(640, 960))
        }.build()

        val preview = Preview(previewConfig)

        preview.setOnPreviewOutputUpdateListener {
            val parent = viewFinder.parent as ViewGroup
            parent.removeView(viewFinder)
            parent.addView(viewFinder, 0)

            viewFinder.surfaceTexture = it.surfaceTexture
            updateTransform()
        }

        val imageCaptureConfig = ImageCaptureConfig.Builder()
            .apply {
                setCaptureMode(ImageCapture.CaptureMode.MIN_LATENCY)
                setFlashMode(cameraFlashMode)
                setTargetResolution(Size(2688, 4032))
            }.build()

        imageCapture = ImageCapture(imageCaptureConfig)
        CameraX.bindToLifecycle(this, preview, imageCapture)
}
```

closed 2019-11-18 by lijialiang

bipinvaylu, 2020-02-05:
Hi, @lijialiang Is this issue fixed for you? If yes, please share your solution. I am facing the same issue with CameraX API.

rai2270, 2020-04-07:
I have the same issue. setFlashMode() with FlashMode.OFF does not turn off.

anilbattini, 2020-12-31:
Since it's closed and no answer is here, let me update you guys.
I see ImageCapture.flashMode only has effect during you build with initial configuration, ImageCapture.Builder() etc.
But if you want to enable/disable flash dynamically, you will have to use the following.

camera?.cameraControl?.enableTorch(enableFlash)

If you are wondering what camera is? Captured it from documentation.

 // A variable number of use-cases can be passed here -
 // camera provides access to CameraControl & CameraInfo
            camera = cameraProvider.bindToLifecycle(
                this, cameraSelector, preview, imageCapture
            )