android/camera-samples #104: CameraX pause video recording

Axrorxoja, 2019-11-20:
Please follow the following instructions before filing a bug:

1. Github issues under this project are only for **sample**-related issues. If you have modified the sample code in any way, then this is probably not the appropriate place to report a bug.
1. If you are trying to report a bug related to the **CameraX library** then you can do so at the Android Issue Tracker: https://issuetracker.google.com/components/618491.
1. Include the specific sample where you found the issue, in addition to device model and specific Android build number -- which can be found under Settings > About phone.
1. Include error logs from logcat, if there are any.
1. If you can, also attach a [bug report](https://developer.android.com/studio/debug/bug-report)-- but please keep in mind that personal information can sometimes be found in those, so double check before attaching.

owahltinez, 2019-12-04:
Video recording is currently not supported by CameraX

closed 2019-12-04 by owahltinez

RamapatiMaurya, 2019-12-07:
We have found a CameraX sample on Github https://github.com/robertlevonyan/CameraXDemo. By using this library we can capture photos and videos. While customizing this sample code i am not able to find any way to pause the video recording even start and stop video recording feature is there.

Any help will be appreciated!!!