android/camera-samples #109: NullPointerException OutputConfiguration.getSurfaces()

shahalRahman, 2019-11-25:
Using camerax_version = '1.0.0-alpha06'

After 
CameraCaptureSession.onConfigured() mState=RELEASING
CameraCaptureSession.onReady() RELEASING
CameraCaptureSession.onClosed() 

I'm getting 
E/AndroidRuntime: FATAL EXCEPTION: LegacyCameraCallback
    Process: com.levelup.synk, PID: 20926
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.hardware.camera2.params.OutputConfiguration.getSurfaces()' on a null object reference
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onCaptureErrorLocked(CameraDeviceImpl.java:2276)
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onDeviceError(CameraDeviceImpl.java:1914)
        at android.hardware.camera2.legacy.CameraDeviceUserShim$CameraCallbackThread$CallbackHandler.handleMessage(CameraDeviceUserShim.java:308)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:201)
        at android.os.HandlerThread.run(HandlerThread.java:65)

This happens very randomly but it definitely happens on exiting the camera fragment.

owahltinez, 2019-11-27:
Sample has been updated to `alpha06`, please reference the latest version and reopen if you find any issues with the sample as-is before modifying any code.

closed 2019-11-27 by owahltinez

shahalRahman, 2019-12-05:
It's crashing on `alpha06` as well.

samohd, 2022-07-19:
Did you ever resolve this?