android/camera-samples #110: Didn't find class "android.view.View$OnUnhandledKeyEventListener"

helang1991, 2019-11-27:
Please follow the following instructions before filing a bug:
11-23 00:18:56.266 I/art     ( 9405): Rejecting re-init on previously-failed class java.lang.Class<androidx.core.view.ViewCompat$OnUnhandledKeyEventListenerWrapper>: java.lang.NoClassDefFoundError: Failed resolution of: Landroid/view/View$OnUnhandledKeyEventListener;
11-23 00:18:56.266 I/art     ( 9405):   at void androidx.core.view.ViewCompat.setOnApplyWindowInsetsListener(android.view.View, androidx.core.view.OnApplyWindowInsetsListener) (ViewCompat.java:2203)
11-23 00:18:56.266 I/art     ( 9405):   at android.view.ViewGroup androidx.appcompat.app.AppCompatDelegateImpl.createSubDecor() (AppCompatDelegateImpl.java:637)
11-23 00:18:56.266 I/art     ( 9405):   at void androidx.appcompat.app.AppCompatDelegateImpl.ensureSubDecor() (AppCompatDelegateImpl.java:518)
11-23 00:18:56.266 I/art     ( 9405):   at void androidx.appcompat.app.AppCompatDelegateImpl.setContentView(int) (AppCompatDelegateImpl.java:466)
11-23 00:18:56.266 I/art     ( 9405):   at void androidx.appcompat.app.AppCompatActivity.setContentView(int) (AppCompatActivity.java:140)
11-23 00:18:56.266 I/art     ( 9405):   at void com.juma.take_picture_plugin.BaseCameraActivity.onCreate(android.os.Bundle) (BaseCameraActivity.java:73)
11-23 00:18:56.266 I/art     ( 9405):   at void com.juma.take_picture_plugin.TakePictureActivity.onCreate(android.os.Bundle) (TakePictureActivity.java:28)
11-23 00:18:56.266 I/art     ( 9405):   at void android.app.Activity.performCreate(android.os.Bundle) (Activity.java:6821)
11-23 00:18:56.266 I/art     ( 9405):   at void android.app.Instrumentation.callActivityOnCreate(android.app.Activity, android.os.Bundle) (Instrumentation.java:1121)
11-23 00:18:56.266 I/art     ( 9405):   at android.app.Activity android.app.ActivityThread.performLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent) (ActivityThread.java:2656)
11-23 00:18:56.266 I/art     ( 9405):   at void android.app.ActivityThread.handleLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:2764)
11-23 00:18:56.266 I/art     ( 9405):   at void android.app.ActivityThread.-wrap12(android.app.ActivityThread, android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:-1)
11-23 00:18:56.266 I/art     ( 9405):   at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:1502)
11-23 00:18:56.266 I/art     ( 9405):   at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:102)
11-23 00:18:56.267 I/art     ( 9405):   at void android.os.Looper.loop() (Looper.java:154)
11-23 00:18:56.267 I/art     ( 9405):   at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6177)
11-23 00:18:56.267 I/art     ( 9405):   at java.lang.Object java.lang.reflect.Method.invoke!(java.lang.Object, java.lang.Object[]) (Method.java:-2)
11-23 00:18:56.267 I/art     ( 9405):   at void com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run() (ZygoteInit.java:933)
11-23 00:18:56.267 I/art     ( 9405):   at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:823)
11-23 00:18:56.267 I/art     ( 9405): Caused by: java.lang.ClassNotFoundException: Didn't find class "android.view.View$OnUnhandledKeyEventListener" on path: DexPathList[[zip file "/data/app/com.jumaps.driver.platform-1/base.apk"],nativeLibraryDirectories=[/data/app/com.jumaps.driver.platform-1/lib/arm64, /data/app/com.jumaps.driver.platform-1/base.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64]]
11-23 00:18:56.267 I/art     ( 9405):   at java.lang.Class dalvik.system.BaseDexClassLoader.findClass(java.lang.String) (BaseDexClassLoader.java:56)
11-23 00:18:56.267 I/art     ( 9405):   at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:380)
11-23 00:18:56.267 I/art     ( 9405):   at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String) (ClassLoader.java:312)
11-23 00:18:56.267 I/art     ( 9405):   at void androidx.core.view.ViewCompat.setOnApplyWindowInsetsListener(android.view.View, androidx.core.view.OnApplyWindowInsetsListener) (ViewCompat.java:2203)
11-23 00:18:56.267 I/art     ( 9405):   at android.view.ViewGroup androidx.appcompat.app.AppCompatDelegateImpl.createSubDecor() (AppCompatDelegateImpl.java:637)
11-23 00:18:56.267 I/art     ( 9405):   at void androidx.appcompat.app.AppCompatDelegateImpl.ensureSubDecor() (AppCompatDelegateImpl.java:518)
11-23 00:18:56.267 I/art     ( 9405):   at void androidx.appcompat.app.AppCompatDelegateImpl.setContentView(int) (AppCompatDelegateImpl.java:466)
11-23 00:18:56.267 I/art     ( 9405):   at void androidx.appcompat.app.AppCompatActivity.setContentView(int) (AppCompatActivity.java:140)
11-23 00:18:56.267 I/art     ( 9405):   at void com.juma.take_picture_plugin.BaseCameraActivity.onCreate(android.os.Bundle) (BaseCameraActivity.java:73)
11-23 00:18:56.267 I/art     ( 9405):   at void com.juma.take_picture_plugin.TakePictureActivity.onCreate(android.os.Bundle) (TakePictureActivity.java:28)
11-23 00:18:56.267 I/art     ( 9405):   at void android.app.Activity.performCreate(android.os.Bundle) (Activity.java:6821)
11-23 00:18:56.267 I/art     ( 9405):   at void android.app.Instrumentation.callActivityOnCreate(android.app.Activity, android.os.Bundle) (Instrumentation.java:1121)
11-23 00:18:56.267 I/art     ( 9405):   at android.app.Activity android.app.ActivityThread.performLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent) (ActivityThread.java:2656)
11-23 00:18:56.267 I/art     ( 9405):   at void android.app.ActivityThread.handleLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:2764)
11-23 00:18:56.267 I/art     ( 9405):   at void android.app.ActivityThread.-wrap12(android.app.ActivityThread, android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:-1)
11-23 00:18:56.267 I/art     ( 9405):   at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:1502)
11-23 00:18:56.267 I/art     ( 9405):   at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:102)
11-23 00:18:56.267 I/art     ( 9405):   at void android.os.Looper.loop() (Looper.java:154)
11-23 00:18:56.267 I/art     ( 9405):   at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6177)
11-23 00:18:56.267 I/art     ( 9405):   at java.lang.Object java.lang.reflect.Method.invoke!(java.lang.Object, java.lang.Object[]) (Method.java:-2)
11-23 00:18:56.267 I/art     ( 9405):   at void com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run() (ZygoteInit.java:933)
11-23 00:18:56.267 I/art     ( 9405):   at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:823)

Device: 
brand:Gionee GN5007
android version: 7.1.1

owahltinez, 2019-11-27:
What sample are you seeing this in? Are you modifying the sample in any way?

owahltinez, 2019-12-23:
Closing due to inactivity. Please reopen if you are still encountering this issue using the latest version of the samples.

closed 2019-12-23 by owahltinez

helang1991, 2020-04-28:
ok, I don't find this issue any more.