android/camera-samples #112: java.util.List android.hardware.camera2.params.OutputConfiguration.getSurfaces()' on a null object reference

Sun-Shuai, 2019-11-29:
I met a NullPointerException during monkey test.

java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.hardware.camera2.params.OutputConfiguration.getSurfaces()' on a null object reference

owahltinez, 2019-12-04:
Thanks for opening this issue. A few questions that can help us fix it:

- What device are you seeing this in?
- What sample are you trying to run?
- Did you modify the sample in any way?

Sun-Shuai, 2019-12-05:
1.Redmi 6 Pro with Android 9
2.I run the CameraXBasic sample
3.I didn't make any changes

owahltinez, 2019-12-05:
Thank you for the additional information. Can you provide us with a more complete stack trace and reproduction steps?

Sun-Shuai, 2019-12-05:
This is log file of monkey test. Hope it can help you.
https://drive.google.com/file/d/1lXUamIDO4cRPK14pU6ifKxbuxenvKbxs/view?usp=sharing

uktshukg, 2019-12-24:
any fix ? need to release to production waiting for this fix

owahltinez, 2020-01-06:
@Sun-Shuai I was not able to find any references to the null pointer exception in the bugreport provided. I am still unable to reproduce the issue performing manual testing.

uktshukg, 2020-01-15:
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.hardware.camera2.params.OutputConfiguration.getSurfaces()' on a null object reference
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onCaptureErrorLocked(CameraDeviceImpl.java:2276)
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onDeviceError(CameraDeviceImpl.java:1914)
       at android.hardware.camera2.legacy.CameraDeviceUserShim$CameraCallbackThread$CallbackHandler.handleMessage(CameraDeviceUserShim.java:308)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loop(Looper.java:201)
       at android.os.HandlerThread.run(HandlerThread.java:65)

uktshukg, 2020-01-15:
@owahltinez

uktshukg, 2020-01-15:
Any hack around, any temporary  solution ?

serbelga, 2020-01-15:
I'm using cameraX in the 1.0.0-alpha06 version and I've had this same problem with this device (Redmi Note 6 Pro) and Xiaomi Mi Mix 2.

soniarojhwal, 2020-01-15:
Hey, 
I am also facing this issue:
Details - 
CameraX version- 1.0.0-alpha05
Device- Xiomi Mi A2 
Scenario- Open and close the camera screen 4-5 times in app

`E/AndroidRuntime: FATAL EXCEPTION: LegacyCameraCallback
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.hardware.camera2.params.OutputConfiguration.getSurfaces()' on a null object reference
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onCaptureErrorLocked(CameraDeviceImpl.java:2276)
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onDeviceError(CameraDeviceImpl.java:1914)
        at android.hardware.camera2.legacy.CameraDeviceUserShim$CameraCallbackThread$CallbackHandler.handleMessage(CameraDeviceUserShim.java:308)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)`

serbelga, 2020-01-15:
It seems that with 1.0.0-alpha08 this issue is solved.

owahltinez, 2020-01-17:
Thanks @serbelga for the update. Closing this issue for now, please reopen if it's not fixed with the latest version of CameraX.

closed 2020-01-17 by owahltinez

uktshukg, 2020-01-17:
@owahltinez   i am on alpha 6 , alpha 6 to 8 requires massive changes because of alpha 7, any hack to do without upgrading the library and fixing this crash?

RobertKDroid, 2020-07-28:
So i managed to keep the whole camera in a fragment and detach it everytime i go out of the screen then attach it onResume which seems to be working

samohd, 2022-07-19:
I am seeing this same issue (with Record Stop/Start repeatedly) with Camera2.

AyakoHoriuchi, 2023-04-05:
This issue occurred in CameraX 1.2.1.

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'android.view.Surface android.hardware.camera2.params.OutputConfiguration.getSurface()' on a null object reference
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onCaptureErrorLocked(CameraDeviceImpl.java:1925)
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onDeviceError(CameraDeviceImpl.java:1693)
       at android.hardware.camera2.legacy.CameraDeviceUserShim$CameraCallbackThread$CallbackHandler.handleMessage(CameraDeviceUserShim.java:299)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:154)
       at android.os.HandlerThread.run(HandlerThread.java:61)

zach-phood, 2023-09-12:
I see this crash occurring at a high rate with CameraX version 1.2.3.

NullPointerException: Attempt to invoke virtual method 'java.util.List android.hardware.camera2.params.OutputConfiguration.getSurfaces()' on a null object reference

What can I do to stop this from crashing my app? What Can I do to stop this from happening? My camera implementation closely resembles the [CameraX code lab.](https://developer.android.com/codelabs/camerax-getting-started#3)

wenhungwww, 2024-03-22:
Thank you for reporting this crash. We're aware of the issue, and it seems like it might stem from an underlying camera HAL problem that can vary depending on the device manufacturer and chipset. While we work as expected, unexpected camera HAL issues can sometimes trigger this error in CameraX.

We're actively investigating to improve the situation. Potential solutions might involve adjustments like delays to mitigate the impact of camera failures. Unfortunately, reproducing the issue reliably in our lab has been difficult.

Could you help?  If you're encountering this crash frequently, could you please provide:

Full device logs leading up to the crash (if possible).
Device model and Android version.
This information would be immensely valuable in our troubleshooting efforts.

Thanks for your patience as we work on this!

pannerr, 2024-04-05:
Hi @wenhungwww I have faced that issue.

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.hardware.camera2.params.OutputConfiguration.getSurfaces()' on a null object reference
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onCaptureErrorLocked(CameraDeviceImpl.java:2121)
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onDeviceError(CameraDeviceImpl.java:1804)
       at android.hardware.camera2.legacy.CameraDeviceUserShim$CameraCallbackThread$CallbackHandler.handleMessage(CameraDeviceUserShim.java:306)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loop(Looper.java:164)
       at android.os.HandlerThread.run(HandlerThread.java:65)
       
       Device model : TC25
      Android version : 8.1.0

samohd, 2024-04-05:
I have reported a similar issue in another post... https://github.com/android/camera-samples/issues/495   I would love to resolve this issue, so anything I can do to assist with fixing it please DM me. It has unfortunately held up my project for some 2 years.

wenhungwww, 2024-04-09:
Hi,

Thank you for reporting this crash. I've been able to reproduce the issue, and it seems to stem from an unhandled CameraDevice error during CameraCaptureSession configuration. This can lead to the legacy Camera2 framework being unable to retrieve the OutputConfiguration, causing the NullPointerException.

We're working on a workaround to mitigate this crash. Due to the variable timing of camera errors, the goal is to cover as many scenarios as possible. I'll keep you updated on our progress and let you know once the changes are ready.

Thank you for your patience and for helping us improve CameraX!

samohd, 2024-04-09:
[wenhungwww](https://github.com/wenhungwww) Can you advise if the solution you're working on is specific to CameraX or if it's with the underlying Camera2? Might your solution be applicable to my post https://github.com/android/camera-samples/issues/495 as well? I had considered reworking my project with CameraX, but my understanding was that CameraX is just a wrapper for Camera2, and thus I'd likely still encounter the same issue. Am I wrong in this thinking?

wenhungwww, 2024-04-10:
[samohd](https://github.com/samohd) Thanks for your question! Here's the latest on my workaround and how it could apply to your situation:

I'm making the change within CameraX code. The root problem lies in the legacy Android Camera2 framework, which legacy devices are unlikely to get updates for.
You're correct that CameraX is built on top of Camera2. While the core issue remains, using CameraX (once the fixed version is released) could resolve it for you.

If you're using Android Camera2 directly, here's the workaround idea:
The issue seems to occur specifically when opening a second CameraCaptureSession. The first session often works fine.  My recommendation is to close the camera device completely after your first CameraCaptureSession, and then reopen it before your second session.

Steps:
- Open CameraDevice
- Open CameraCaptureSession (1st)
- Close CameraDevice (wait for the 'onClosed' callback)
- Reopen CameraDevice
- Open CameraCaptureSession (2nd)

This forces a reset of the legacy camera's internal state, which can prevent the getSurface() null issue.

Important Notes:
This isn't a perfect solution. Other errors are possible, and device-specific changes by manufacturers could affect its success.

samohd, 2024-04-10:
[wenhungwww](https://github.com/wenhungwww) The root problem lies in the legacy Android Camera2 framework, which legacy devices are unlikely to get updates for.

Does this suggest that a newer camera device might already have a fix for this, and thus if I use a newer device it might already work as is? Or do newer devices still have the same Camera2 base?

My process is essentially this:

Camera_Open / Transform
loop:
  previewSession Create
  MediaRecorer configure
  MediaRecorder.Start
  MediaRecorder.StopRepeating
  MediaRecorder.AbortCaptures
  MediaRecorder.Stop
  MediaRecorder.Reset
  previewSession Close
goto loop

The crash consistently happens after MediaRecorder.Stop which seems unrelated to the previewSession being closed and reopened, which it doesn't get to. Usually it gets thru several loops before it fails - sometimes many loops. I rarely see it fail at only the second loop.

MediaRecorderJNI] stop
[Binder] Caught a RuntimeException from the binder stub implementation.
[Binder] java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.hardware.camera2.params.OutputConfiguration.getSurfaces()' on a null object reference

I'm not sure what the relationship is between MediaRecorder.stop and getSurfaces(). I never call getSurfaces() in my code. Perhaps it's rebuilding the onscreen preview without the recording portion.

So, I'm not sure if your solution can resolve how I see it play out... as it is going to crash on the MediaRecorder.stop call... so I can't get back to a point where I close the camera and open it again for the next record session.

samohd, 2024-04-10:
Also note that, what I've observered is that the idea of opening a camera app, starting a recording, stopping a record, and repeating that endlessly... never causes a crash on any of the device's native camera apps. Which suggests there's a way for this to work with Android Camera code... or that the manufacturers are not using the Android Camera code to do this.

Thoughts?

samohd, 2024-04-10:
What is your sceneraio in which you are Opening a CameraCaptureSession multiple times, to result in it crashing? Are you also doing recording?

samohd, 2024-04-10:
I also can't try/catch the crash and work around it (close and reopen camera), as try/catch misses it (likely because it's not my code that's crashed, but underlying Camera2 code.

When I run debug from Visual Studio, it will halt at "cameraDevice.CreateCaptureSession(surfaces, new CameraCaptureStateCallback(this), backgroundHandler);"

With the error:
Android.Hardware.Camera2.CameraAccessException
  Message=CAMERA_ERROR (3): waitUntilIdle:1701: Camera 0: Error waiting to drain: Connection timed out (-110)

But this code halt occurs AFTER the crash that gets shown in the Output log in Visual Studio, following the STOP. And thus this error at the debug code halt is occurring when it's already moved on to be ready before the next record stop. Which is to say, the debugger isn't even aware the crash error has happened either,  just as with try/catch not kicking in.

wenhungwww, 2024-04-11:
[samohd](https://github.com/samohd) Here's a breakdown to clarify the situation:

1. Legacy Camera Code:
You're correct that newer devices might have fixes in their Camera2 implementation. You can check this using the INFO_SUPPORTED_HARDWARE_LEVEL property (https://developer.android.com/reference/android/hardware/camera2/CameraCharacteristics#INFO_SUPPORTED_HARDWARE_LEVEL).
Legacy Camera2 code exists across multiple Android API levels, even on devices receiving updates.
2. Root of the Issue:
The core problem starts within the camera hardware abstraction layer (cameraHAL). There could be multiple triggers (capture pipeline instability, sensor issues, memory issues, etc.).
When the camera device reports an error, the legacy framework layer sometimes encounters a null pointer issue when trying to find the output configuration. This often surfaces when configuring a second CameraCaptureSession.
My workaround is preventative: closing the camera device fully between capture sessions aims to reset the camera's state and avoid this null pointer scenario.
3. Issue Frequency & Workarounds:
The issue is low probability but persists. Two approaches to mitigate it:
Prevention: Ideally, prevent the initial camera device error. This is difficult, as causes can be device-specific and hard to pinpoint.
Workaround: Proactively clearing the output configuration mitigates the null pointer issue, which is the focus of my workaround.
4. Exception Handling:
You're right, the null pointer occurs in a background camera thread, making try/catch ineffective. Thread.setDefaultUncaughtExceptionHandler is an option, but not ideal for this.
5. Ongoing Challenges
My focus is on preventing the null pointer crash. The camera device error itself may have other roots. We'd need a separate discussion focusing on the causes of that initial error, which could be incorrect operations, device instability, and more.

samohd, 2024-04-12:
[wenhungwww](https://github.com/wenhungwww)
1. The Galaxy S9 I was/am developing on reports back as INFO_SUPPORTED_HARDWARE_LEVEL_3 which appears to be the highest feature set. I have also tested on S10 and Pixel 4, which both have the same crash.

I was endeavoring to keep the hardware cost down on this project, with consideration that I need 6-10 devices for each implementation, and at least 240fps capture. There's aso a potential of damage to the cameras in the process of the use. Which is to say, I don't ideally want to just go out and buy the latest camera each year, in hopes that I'll finally finally a model that works reliably.

With all that in mind, are you aware of any 240fps capable devices on which this crash does not happen?

3. Do you think that this process could potentially work around the crash:
loop:
Camera_Open
previewSession Create
MediaRecorder.Start
MediaRecorder.Stop
previewSession Close
Camera_Close
goto loop

This could involve a fair amount of rework of the code to get it to the point of testing if the Camera_Close/Reopen works.

Note that when I was actively developing the project, two years ago, in my testing I very rarely saw the crash. But, other testers who were testing on multiple devices at once, saw it in almost every iteration on at least one of the devices. It wasn't until I wrote in some test code to loop record start/stop endlessly, that I was able to repeatedly see it. For our record start/stop in a loop scenario, it happens regularly.

4. "Proactively clearing the output configuration"... I presume you mean by doing Camera_Close after recording.

5. Given that, as of this point, I've had to abandon the project completly, after putting a ton of work into it. I'd love to get the project up and running again. So, if there's an issue that'll require fixes beyond what I can do in my code, I'm certainly interested to participate in that exploration. If you want to take this discussion off github, I'm happy to do that as well. I don't see anyway on github to even private message you.