android/camera-samples #114: CameraXBASIC not working on Xiaomi REDMI 3S

Kashonkov, 2019-12-03:
Hi!

When i had tried to install and run sample app  on xiaomi i received next Error:

com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.cameraxbasic, PID: 25177
    java.lang.ExceptionInInitializerError
        at androidx.camera.core.Preview.<clinit>(Preview.java:84)
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder.<init>(AutoFitPreviewBuilder.kt:91)
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder.<init>(AutoFitPreviewBuilder.kt:39)
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder$Companion.build(AutoFitPreviewBuilder.kt:243)
        at com.android.example.cameraxbasic.fragments.CameraFragment.bindCameraUseCases(CameraFragment.kt:283)
        at com.android.example.cameraxbasic.fragments.CameraFragment.access$bindCameraUseCases(CameraFragment.kt:92)
        at com.android.example.cameraxbasic.fragments.CameraFragment$onViewCreated$1.run(CameraFragment.kt:253)
        at android.os.Handler.handleCallback(Handler.java:742)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:157)
        at android.app.ActivityThread.main(ActivityThread.java:5571)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:745)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:635)
     Caused by: java.lang.IllegalStateException: CameraX not initialized yet.
        at androidx.camera.core.CameraX.getCameraDeviceSurfaceManager(CameraX.java:697)
        at androidx.camera.core.CameraX.getSurfaceManager(CameraX.java:454)
        at androidx.camera.core.Preview$Defaults.<clinit>(Preview.java:641)
        at androidx.camera.core.Preview.<clinit>(Preview.java:84) 
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder.<init>(AutoFitPreviewBuilder.kt:91) 
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder.<init>(AutoFitPreviewBuilder.kt:39) 
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder$Companion.build(AutoFitPreviewBuilder.kt:243) 
        at com.android.example.cameraxbasic.fragments.CameraFragment.bindCameraUseCases(CameraFragment.kt:283) 
        at com.android.example.cameraxbasic.fragments.CameraFragment.access$bindCameraUseCases(CameraFragment.kt:92) 
        at com.android.example.cameraxbasic.fragments.CameraFragment$onViewCreated$1.run(CameraFragment.kt:253) 
        at android.os.Handler.handleCallback(Handler.java:742) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at android.os.Looper.loop(Looper.java:157) 
        at android.app.ActivityThread.main(ActivityThread.java:5571) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:745) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:635)  

What can I do with this?
thanks

zhaonian, 2019-12-04:
I'd really like to help but I don't have a RedMI to repro this. I wonder if inlining the initialization of Preview could help :/ 
Can you please try this maybe?
https://github.com/zhaonian/camera-samples/tree/fix/initialization-exception

Kashonkov, 2019-12-04:
Thanks, I'll try it and will say if it will be fine

Kashonkov, 2019-12-04:
so, it's not working, It seems that Camera2Initializer doesnt initialize on ths device (i couldn't see that methode onCreate called, although  on Samsung s8+  it was)

owahltinez, 2019-12-23:
Hey @Kashonkov are you modifying the sample in any way? We have updated the way CameraX is initialize in `alpha07`, then again in `alpha08`. There will likely be some further minor tweaks prior to beta release. Perhaps the new initialization code in the latest version of the library helps.

Kashonkov, 2019-12-23:
Hi, Nope Id didn't modify it, Let mi try the new version tomorrow, I will write did it help or not

Kashonkov, 2019-12-26:
Yes, It's working. Thanks!

closed 2019-12-26 by Kashonkov