android/camera-samples #115: White screen in device with multiple cameras

chill-cow, 2019-12-04:
I launched this app in a device with multiple cameras and camera preview becomes white. But when I click the button to take a picture the ImageCupture provides me a correct image.

I tested it in a few devices with multiple cameras and got the same issue.

owahltinez, 2019-12-23:
Thanks for opening this issue. A few questions that can help us fix it:

* What device are you seeing this in?
* What sample are you trying to run?
* Did you modify the sample in any way?

owahltinez, 2020-01-24:
Closing due to inactivity. Please reopen if this is still an issue.

closed 2020-01-24 by owahltinez

pbsarathy21, 2020-03-06:
Me too face the same issue in Android pie.

On opening Camera shows white screen, if i go to home screen and come back with recent activities, then it works perfectly