android/camera-samples #116: Captures photo once, then never again, even after reinstall

dyaeger12345, 2019-12-05:
After installing the BasicJava example on my Android 9.0 device (Moto G7 Supra), I was able to capture one photo. All subsequent capture attempts resulted in no image being saved and resulted in the following log message:

2019-12-05 16:08:50.545 26851-26920/com.example.android.camera2basic W/CameraCaptureSession: Session 0: abortCaptures - Session is already aborting; doing nothing

This continues to occur even after uninstalling and reinstalling the app. It seems as if the Camera service is never released.

owahltinez, 2020-01-06:
I am unable to reproduce the issue. It sounds like a device-specific problem where the camera did not close correctly. What happens if you restart the device?

owahltinez, 2020-02-06:
Closing due to inactivity

closed 2020-02-06 by owahltinez