android/camera-samples #124: Capture picture once and after that does not work until i remove that app from backgound , camera2basicJava

asifk1997, 2019-12-17:
Please follow the following instructions before filing a bug:

1. Here is a link of video in which i have shown whats happening https://drive.google.com/open?id=1JnhdEDm34EW5Nt5x4jfLqDZnEqdbggq2
1. So for the first time I am clicking a picture its working but after subsequent clicks its not showing anything or clicking picture.
1. Phone Moto G4 Plus
1. logs: 
2019-12-17 13:26:50.729 30380-30398/com.example.android.camera2basic I/OpenGLRenderer: Initialized EGL, version 1.4
2019-12-17 13:26:50.729 30380-30398/com.example.android.camera2basic D/OpenGLRenderer: Swap behavior 1
2019-12-17 13:26:50.742 30380-30380/com.example.android.camera2basic I/CameraManagerGlobal: Connecting to camera service
2019-12-17 13:26:50.781 30380-30380/com.example.android.camera2basic I/CameraManager: Using legacy camera HAL.
2019-12-17 13:26:50.932 30380-30397/com.example.android.camera2basic I/CameraDeviceState: Legacy camera service transitioning to state CONFIGURING
2019-12-17 13:26:50.933 30380-30446/com.example.android.camera2basic I/RequestThread-0: Configure outputs: 2 surfaces configured.
2019-12-17 13:26:50.933 30380-30446/com.example.android.camera2basic D/Camera: app passed NULL surface
2019-12-17 13:26:50.951 30380-30446/com.example.android.camera2basic I/RequestThread-0: configureOutputs - set take picture size to 4608x3456
2019-12-17 13:26:50.978 30380-30397/com.example.android.camera2basic I/CameraDeviceState: Legacy camera service transitioning to state IDLE
2019-12-17 13:26:50.990 30380-30397/com.example.android.camera2basic I/RequestQueue: Repeating capture request set.
2019-12-17 13:26:50.999 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-12-17 13:26:50.999 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:50.999 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:51.402 30380-30449/com.example.android.camera2basic I/CameraDeviceState: Legacy camera service transitioning to state CAPTURING
2019-12-17 13:26:55.022 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-12-17 13:26:55.022 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:55.022 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:55.066 30380-30399/com.example.android.camera2basic D/LegacyFocusStateMapper: onAutoFocusMoving - ignoring move callbacks from old af run1
2019-12-17 13:26:55.087 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-12-17 13:26:55.087 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:55.087 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:55.087 30380-30397/com.example.android.camera2basic I/RequestQueue: Repeating capture request cancelled.
2019-12-17 13:26:55.088 30380-30397/com.example.android.camera2basic I/RequestThread-0: Flushing all pending requests.
2019-12-17 13:26:55.088 30380-30397/com.example.android.camera2basic E/RequestQueue: cancel failed: no repeating request exists.
2019-12-17 13:26:55.116 30380-30446/com.example.android.camera2basic I/CameraDeviceState: Legacy camera service transitioning to state IDLE
2019-12-17 13:26:55.122 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-12-17 13:26:55.122 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:55.122 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:55.124 30380-30385/com.example.android.camera2basic I/art: Do partial code cache collection, code=22KB, data=29KB
2019-12-17 13:26:55.126 30380-30385/com.example.android.camera2basic I/art: After code cache collection, code=22KB, data=29KB
2019-12-17 13:26:55.126 30380-30385/com.example.android.camera2basic I/art: Increasing code cache capacity to 128KB
2019-12-17 13:26:55.211 30380-30399/com.example.android.camera2basic I/CameraDeviceState: Legacy camera service transitioning to state CAPTURING
2019-12-17 13:26:55.546 30380-30399/com.example.android.camera2basic I/RequestThread-0: Received jpeg.
2019-12-17 13:26:55.548 30380-30399/com.example.android.camera2basic I/RequestThread-0: Producing jpeg buffer...
2019-12-17 13:26:55.608 30380-30446/com.example.android.camera2basic I/CameraDeviceState: Legacy camera service transitioning to state IDLE
2019-12-17 13:26:55.611 30380-30397/com.example.android.camera2basic D/Camera2BasicFragment: /storage/emulated/0/Android/data/com.example.android.camera2basic/files/pic.jpg
2019-12-17 13:26:55.612 30380-30397/com.example.android.camera2basic I/RequestQueue: Repeating capture request set.
2019-12-17 13:26:55.618 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-12-17 13:26:55.618 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:55.618 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:55.643 30380-30449/com.example.android.camera2basic I/CameraDeviceState: Legacy camera service transitioning to state CAPTURING
2019-12-17 13:26:55.670 30380-30399/com.example.android.camera2basic D/LegacyFocusStateMapper: onAutoFocusMoving - ignoring move callbacks from old af run2
2019-12-17 13:26:55.687 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-12-17 13:26:55.687 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:26:55.687 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:27:00.458 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-12-17 13:27:00.458 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:27:00.459 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:27:00.499 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-12-17 13:27:00.499 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:27:00.499 30380-30446/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-12-17 13:27:04.062 30380-30380/com.example.android.camera2basic I/RequestQueue: Repeating capture request cancelled.

owahltinez, 2020-01-06:
To rule out whether it is an issue with the device, can you confirm if the Camera2Formats sample works well on your device?

asifk1997, 2020-01-12:
Camera2Formats, CameraXBasic app was working fine but i have seen this issue in 2 phones Moto G4 plus and redmi Go

asifk1997, 2020-01-12:
2020-01-12 15:35:33.948 20170-20237/com.example.android.camera2basic W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2020-01-12 15:35:33.948 20170-20237/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2020-01-12 15:35:33.948 20170-20237/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2020-01-12 15:35:33.985 20170-20237/com.example.android.camera2basic W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2020-01-12 15:35:33.986 20170-20237/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2020-01-12 15:35:33.986 20170-20237/com.example.android.camera2basic W/LegacyRequestMapper: Only received metering rectangles with weight 0.

asifk1997, 2020-01-12:
logs were showing like these after i clicked on take picture

owahltinez, 2020-01-17:
Please use Camera2Formats as a reference, we will be deprecating Camera2Basic in the near future

closed 2020-01-17 by owahltinez