android/camera-samples #128: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.media.MediaCodec.queueInputBuffer(int, int, int, long, int)' on a null object reference

williamycyh, 2019-12-19:
java.lang.NullPointerException: Attempt to invoke virtual method 'void android.media.MediaCodec.queueInputBuffer(int, int, int, long, int)' on a null object reference
--
1 | at androidx.camera.core.VideoCapture.a(SourceFile:147)
2 | at androidx.camera.core.VideoCapture$1.run(SourceFile:1)
3 | at android.os.Handler.handleCallback(Handler.java:739)
4 | at android.os.Handler.dispatchMessage(Handler.java:95)
5 | at android.os.Looper.loop(Looper.java:148)
6 | at android.os.HandlerThread.run(HandlerThread.java:61)

I got an error with 1.0.0-alpha06 when take video, I have no idea to resolve ? pls help.

yasintanriverdi, 2019-12-19:
Getting the same error with the same version when the fragment is destroyed during video recording.

williamycyh, 2019-12-20:
> Getting the same error with the same version when the fragment is destroyed during video recording.

how to resolve it, i got a lot of errors on my online app....

yasintanriverdi, 2019-12-20:
Unfortunately, could not find any solution

owahltinez, 2019-12-23:
Thanks for opening this issue. A few questions that can help us fix it:

* What device are you seeing this in?
* What sample are you trying to run?
* Did you modify the sample in any way?

williamycyh, 2019-12-24:
> * Did you modify the sample in any way?
I am modify the demo to take video.
I think, this bug can be reproduced on all devices, start record video, without stopping the video recording, just exit the fragment will appear

owahltinez, 2020-01-06:
This repo is used to track issues with the samples. If you have any questions related to the usage of the library, please use StackOverflow (tag `android-camerax`) or start a thread in the [developer forums](g.co/camerax/developers).

closed 2020-01-06 by owahltinez

lhhuu, 2020-10-16:
camera X is not good