android/camera-samples #137: Camera Preview from phone has camera 5MP is brighter than the other one has camera 8MP

namphho, 2019-12-26:
Hi friends, 
I have 2 Android PDA (android 6.0). The one has a camera 5MP and the other one have a camera 8MP.
My application use Camera API v2 to take a picture. So I'm facing an issue that Camera Preview of 5MP phone is
brighter than the 8MP phone. I expected that 8MP one is brighter than the 5MP one. I have 2 questions:
1/ is Camera Api v2 incompatible with the camera 8MP module? Quality of camera 8MP is bad
2/ if the root cause is that Camera API v2 is incompatible, so is there any workaround? 
any advice and suggestions will be greatly appreciated

Photo from the camera 5 MP: https://imgur.com/4UXZm9b
Photo from the camera 8 MP: https://imgur.com/GPPhYdF

owahltinez, 2020-01-06:
This sounds like an issue with the API themselves, or potentially the device drivers. Please open an issue in the camera component of the [AOSP issue tracker](https://issuetracker.google.com) so we can assign it to the correct team. Please post a link here once the new issue is open, so we can close this one out.

owahltinez, 2020-02-15:
Closing due to inactivity.

closed 2020-02-15 by owahltinez