android/camera-samples #138: Camera2SlowMotion not working properly in many devices

Ashutosh-Tiwari, 2019-12-27:
I installed the Camera2SlowMotion sample in Redmi Note 7 Pro and opened to see a white screen with list of available camera configs and chose one, and then the screen turned complete black and nothing happens then after.

Apart from this I also combined this code with some code from Camera2VideoKotlin and it worked just well for many of the devices. I could very well record slow motion videos with 120 and above fps. But again, when I install the app in Redmi note 7 pro. The app closes at preview. Neither I see any crash log in the logcat nor I am able to catch the crash in a try...catch. It seems to be something non-fatal. But at the end, I am clueless on what is happening here, please help.

TheMelody, 2019-12-31:
Whether to turn on the camera permissions？

TheMelody, 2019-12-31:
You can see if there are other applications that occupy the right to use the camera, and check that the system camera is occupied

Ashutosh-Tiwari, 2019-12-31:
That's not the case I am sure. My research has led me to believe that this is about MI devices having camera 2 API disabled. Or else, I aam still blank.

Ashutosh-Tiwari, 2020-01-02:
@owahltinez  I have noticed that the high speed preview and high speed video capture works fine in devices having the hardware support of `CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_FULL` but the app fails to load the high speed preview in devices having hardware support of `CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_3`
This is weird because the documentation for `CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL` says this:

[image]]

Which means that the devices having hardware support of `CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_3` are supposed to have all the capabilities that are there in devices having hardware support of `CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_FULL`.

Ashutosh-Tiwari, 2020-01-02:
I opened my sample in Galaxy S9 which has hardware support upto CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_3 in camera 2 api and the high speed preview worked, slow motion video capture worked. The same sample, I opened in Redmi note 7 pro which also has hardware support upto CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_3 in camera 2 api but in this device the app crashes at preview with errors like in this stack link: https://stackoverflow.com/questions/55990262/access-denied-for-property-vendor-camera-aux-packagelist

Also to notice that the same app works superb in Galaxy S9 but fails to load the preview and closes in Galaxy S8.

owahltinez, 2020-01-06:
Thanks for reporting this. I don't have a Redmi Note 7 Pro handy, but I will investigate using a Galaxy S8.

Ashutosh-Tiwari, 2020-01-07:
Please.. any figuring out the issue with high speed preview will be a big help. So far I could only narrow down the culprit line to this issue and that is,

settting the fps range for ex. [30, 120] like this `set(CaptureRequest.CONTROL_AE_TARGET_FPS_RANGE, Range(FPS_PREVIEW_ONLY, args.fps))` in the preview request builder is causing the app to close. If I don't set it, the app doesn't close and the preview stays blacked.

Apart from this the change in my code is only a combination of Camera2VideoKotlin and Camera2SlowMotion that takes TextureView instread of SurfaceView for camera handling and so I am also using the methods like `configureTransform()` and I am preparing my previewSurface like this:

    //This is the output Surface we need to start preview.
    private val previewSurface: Surface by lazy {
        Surface(textureViewCamera.surfaceTexture.apply {
            setDefaultBufferSize(previewSize.width, previewSize.height)
        })
    }

owahltinez, 2020-01-13:
@Ashutosh-Tiwari I am working on a commit that will fix the black screen issue for Samsung Galaxy S8, but I can't reproduce the crashing. Can you please do the following:

1. Run `adb logcat -c` to clear your logs
2. Run the sample, watch it crash
3. Collect a bug report following [these instructions](https://developer.android.com/studio/debug/bug-report)
4. Reply to this comment and attach your bug report

Ashutosh-Tiwari, 2020-01-13:
@owahltinez  Thanks for the time you're spending on the issue. Here is one more weird scenario that is happening in Galaxy Note 5 that has LEVEL_FULL hardware support and can record videos at 120fps:

https://drive.google.com/file/d/1k0IkaZH4eVNJgHg-cJ_8nitZh74x_yuJ/view?usp=sharing

The video in the above link is a recording session of testing the video record in the Camera2SloMotion sample. You'll notice that the preview started just fine and the camera was moving too but as soon as I started to record the video with the same movement, the camera movement stopped and video recording become went still within a second. Don't know what went wrong. But this was just to give you an insight so that you can check for the same when you're spending time on the fixes.

owahltinez, 2020-01-13:
Thanks Ashutosh, I think that may be a different issue. I have seen it intermittently in other devices, so I'll take a look. Can you please provide the bug report for both the crash and this new issue?

Ashutosh-Tiwari, 2020-01-15:
Didn't get you on the bug report. Do you want me to share the error/crash log or something?

Ashutosh-Tiwari, 2020-01-15:
@owahltinez  Here is a bug report:

https://drive.google.com/file/d/1j_yJJJVFQjXD6ga3fhzvPADoee5KTVGq/view?usp=sharing

I found the app crashing in a random case for Galaxy Note 5. Don't know if it really has something to do with the above-mentioned issues.

owahltinez, 2020-01-16:
@Ashutosh-Tiwari I am unable to reproduce the crash, but I was able to fix the black screen issue on a Samsung Galaxy S8. Can you please clone [this branch](https://github.com/android/camera-samples/tree/slow-motion-fix) and verify that if fixes your issue as well?

Ashutosh-Tiwari, 2020-01-16:
Hey @owahltinez I don't see the issues fixed. Here is a recording for two devices:

Galaxy Note 5: [Crashed randomly]
https://drive.google.com/file/d/1lLVmmT9fX_smTtq81mvu03rdBSrQ-AL3/view?usp=sharing

Redmi Note 7 Pro: [Blacked preview - tried for all the available camera configs]
https://drive.google.com/file/d/1QE9k107avzzKyM9oJxNsln9aAsbupQyC/view?usp=sharing

owahltinez, 2020-01-16:
Thanks @Ashutosh-Tiwari for your help debugging this. Can you please:
1. Confirm you are using the slow-motion-fix branch I linked above (not master)
2. Provide a bugreport for the Redmi Note 7 Pro device after running this sample

Ashutosh-Tiwari, 2020-01-16:
I'll prepare a bug report and share it with you. And yes, I've made sure I am running the latest code. The code contains all the changes for `overlay` and `@JvmOverloads constructor`, etc.

owahltinez, 2020-01-20:
Pushed the changes since they fix the issue for Samsung S8 devices. Reopening issue until it's fixed in the other reported devices.

owahltinez, 2020-01-28:
I tested this sample app on a RedMi Note 8 and it worked without issues. Please reopen if you can still reproduce and can provide a bugreport.

closed 2020-01-28 by owahltinez

zokica8, 2022-04-06:
Hi. I still have the issue with running this example of capturing a slow motion video on my phone, which is Mi 11T Pro. When I created the bug report for the whole device, I found two lines that could be the reason why this is not working.

ServiceManager: Permission failure: android.permission.SYSTEM_CAMERA from uid=somenumber pid=somenumber
ServiceManager: Permission failure: [android.permission.CAMERA](https://developer.android.google.cn/reference/android/Manifest.permission#CAMERA)_OPEN_CLOSE_LISTENER from uid=somenumber pid=somenumber

On Android 10 and lower devices it works, from Android 11 it does not work.
The only thing that I found was that from Android 11, the OS introduced the system camera permission, which is assigned at the system level. This is from the Android documentation:

"On devices running Android 11 or higher, the Android framework supports system cameras, which are camera devices that are visible only to processes with the android.permission.SYSTEM_CAMERA permission and regular camera permissions. The android.permission.SYSTEM_CAMERA permission, introduced in Android 11, has a protection level of system|signature. This means that only apps installed on the system partition with or signed with the same certificate as the system can be granted this permission. Third-party public apps can't access system cameras.

System cameras are useful for device manufacturers that want to implement features that require access to a camera but are also restricted to privileged or system apps.

Because systems apps that are granted the android.permission.SYSTEM_CAMERA permission must also have the android.permission.CAMERA permission, users can choose to revoke normal CAMERA permissions to prevent such an app from accessing the cameras on the device."

Basically what this means is that the only way you can create this kind of an app and run this example is for this app to be a system app, an app that is part of a system, like most Google apps for most Android phones that you can not uninstall from the device, unless you want to root the device.

Is there anybody who has experience in creating these kind of apps?
Also, I found another link that could be useful.

https://blog.csdn.net/KLYAZWS/article/details/119376225

commit 95969bb8 (2020-01-20): Properly override custom view constructor

- Fixes #138

Change-Id: Iae8c59c4c487dfbc2fe35d3c6504386a70f335d7