android/camera-samples #144: CameraXBasic photos deleted on uninstall; proper Scoped Storage support

mark-ha, 2020-01-12:
When CameraXBasic is uninstalled, any photos taken with it are deleted. I realise this is due to MainActivity.getOutputDirectory() choosing an application-specific directory. I assume this was also done as a workaround for Android 10's scoped storage, but for a camera app example I feel this isn't really appropriate - anyone following this to write a CameraX application may not realise they're creating a camera application that deletes a user's photos when uninstalled, which could be disastrous!

More generally, given Google's push to move us to non-File APIs, it seems inconsistent and mixed messaging not to be showcasing Google's best practices for saving files on Android 10 with scoped storage.

The camera sample should show the correct best practice for saving a photo (which Google documents as using MediaStore API, although as far as I can tell documentation for how to do that seems lacking, so having this shown in the camera sample would be all the more valuable please).

Looking at other samples' source code, it seems this issue may apply to them too (e.g., Camera2BasicJava), but I've only tested CameraXBasic on my device (Galaxy S10e, Android 10).

Thanks!

owahltinez, 2020-01-13:
Thanks Mark for reporting this issue. This definitely requires some thinking on our part. The current behaviour is not a workaround, it is fully intentional: once the app goes away then all the data associated with it should too. It also simplifies testing, otherwise developers would have to manually delete all photos to go back to an empty gallery as opposed to simply uninstalling and reinstalling.

We are willing to consider using scoped storage, although I disagree that using the app's private folder is inconsistent and mixed messaging -- in fact it encourages developers to implement better privacy, not allowing other apps to access the user photos.

In the meantime, for samples related to usage of scoped storage, take a look at: https://github.com/android/storage-samples

mark-ha, 2020-01-13:
Thanks for your reply. That makes about the reasons for when deleting photos upon uninstall is desired. I would still love to see this sample include code to show best practices for when it is desired to have photos remain after uninstall (e.g., "real world" typical third party camera application).

Thanks for the link to the storage-samples examples - do you know if there one that shows saving an image using MediaStore API (the MediaStore sample suggests it's for viewing images in the mediastore rather than adding an image)?

owahltinez, 2020-01-14:
My understanding is that you don't need to interact with the MediaStore APIs for inserting media, in fact that is probably forbidden. Instead, developers should use the MediaScanner APIs, which we already do [in this sample](https://github.com/android/camera-samples/blob/3d1a254eb018a51ff39ae78d39a9e9e7942a027b/CameraXBasic/app/src/main/java/com/android/example/cameraxbasic/fragments/CameraFragment.kt#L219). Pictures taken with this sample already appear in the MediaStore, here's how you can test it:

1. Build and install CameraXBasic sample app
2. Take photo using CameraXBasic sample app
3. Build and install [MediaStore sample app](https://github.com/android/storage-samples/tree/master/MediaStore)
4. Open MediaStore sample app

Then, the photo taken using CameraXBasic sample app appears in the list.

mark-ha, 2020-01-14:
Although would that method work with scoped storage if saving the photo outside the app specific folder? I use that method (File API with MediaScannerConnection.scanFile) currently when saving photos  to a folder inside DCIM, but such photos fail to save when scoped storage is enabled.

I was following https://developer.android.com/training/data-storage which for "Shareable media files (images, audio files, videos)" and "Files removed on app uninstall?=No" says to use MediaStore API.

owahltinez, 2020-01-14:
Scoped storage is more about reading than about writing. For writing files that can be shared with other apps, you have two main options: using a FileProvider or external storage. This sample implements both; the FileProvider for direct-share (from gallery) and external storage to save images. We are using a scoped version of external storage to avoid asking for permissions unnecessary for the app to function.

It's hard to say what could be going wrong with your DCIM folder without more details. Since the issue you just mentioned is not a bug in the sample, I would suggest creating a StackOverflow question with more details (feel free to post the link here).

owahltinez, 2020-03-30:
Closing due to inactivity.

closed 2020-03-30 by owahltinez

genijmath, 2020-04-09:
Oscar,

MediaScanner (the flow you described in the comment on Jan 13) was broken by "Update CameraX to beta1" commit from Feb 27, 2020 (e964394).

It replaced
MediaScannerConnection.scanFile(context, arrayOf(photoFile.absolutePath), ...
with
MediaScannerConnection.scanFile(context, arrayOf(savedUri.toString()), ...

After this change photos do not show up in MediaStore sample.

Log: 
Image capture scanned into media store: null.

Is it a bug in MediaScannerConnection?

owahltinez, 2020-04-12:
It's possible that MediaScannerConnection expects a file path, not a URI. Let me look into this, thanks for flagging.