android/camera-samples #145: question about Camera2/CameraX data size

wangshengyang1996, 2020-01-16:
i'm confused why YUV_420_888 format image's plane-U and plane-V size is an odd number, and data length ratio is about 2: 1: 1（YUV420 data's Y:U:V is supported to be  4:1:1）
```
        val preview = Preview(previewConfig)

        val imageAnalysis = ImageAnalysis(imageAnalysisConfig)
        imageAnalysis.setAnalyzer(executor, ImageAnalysis.Analyzer { image, rotationDegrees ->
            val strBuilder = StringBuilder()
            for (plane in image.planes) {
                strBuilder.append(plane.buffer.capacity())
                    .append(" ")
            }
            Log.i(
                "imgSize",
                "imageFormat is ${image.format}, imageWidth is ${image.width} , imageHeight is ${image.height}, rotationDegrees is ${rotationDegrees}, size is ${strBuilder}"
            )
        })
```
when preview size is 1280x720
` I/imgSize: imageFormat is 35, imageWidth is 1280 , imageHeight is 720, rotationDegrees is 90, size is 921600 460799 460799 `
when preview size is 1920x1080
`I/imgSize: imageFormat is 35, imageWidth is 1920 , imageHeight is 1080, rotationDegrees is 90, size is 2073600 1036799 1036799 `

wangshengyang1996, 2020-01-16:
the function ` getLockedImageInfo` in `frameworks/base/media/jni/android_media_Utils.cpp` shows how to calculate planeU and planeV 's length , but i don't know why the final row's data need to be specially processed.
https://cs.android.com/android/platform/superproject/+/master:frameworks/base/media/jni/android_media_Utils.cpp;l=118;bpv=0;bpt=1
```
 switch (fmt) {
        case HAL_PIXEL_FORMAT_YCbCr_420_888:
            pData =
                (idx == 0) ?
                    buffer->data :
                (idx == 1) ?
                    buffer->dataCb :
                buffer->dataCr;
            // only map until last pixel
            if (idx == 0) {
                pStride = 1;
                rStride = buffer->stride;
                dataSize = buffer->stride * (buffer->height - 1) + buffer->width;
            } else {
                pStride = buffer->chromaStep;
                rStride = buffer->chromaStride;
                dataSize = buffer->chromaStride * (buffer->height / 2 - 1) +
                        buffer->chromaStep * (buffer->width / 2 - 1) + 1;
            }
            break;

```

owahltinez, 2020-01-16:
Hi @wangshengyang1996, this repo is used to track issues with the sample. For questions or issues with the libraries used by the sample, please use StackOverflow or reach out to the developer forums at https://g.co/camerax/developers.

closed 2020-01-16 by owahltinez