android/camera-samples #147: Cannot choose available preview resolution through setTargetResolution() on CameraXBasic

asifk1997, 2020-01-18:
I was trying to use cameraxbasic app on Camera HARDWARE_LEVEL_LEGACY phones so the preview size for imageAnalysis was low resolution so i manually calculated size myself by scalerstreamconfiguration map and preview size closest to screen resolution .
Suppose i chose the preview size on Xolo Era phone and it was 960 : 540.

I gave this size to preview and imageAnalyzer method through setTargetResolution function .
And it was selecting something like 864 : 480.

I want to choose a high preview size for imageAnalyzer like 1280: 720 .

asifk1997, 2020-01-18:
I used this code for best preview size calculation:

CameraManager cameraManager = (CameraManager) MainActivity.this.getSystemService(Context.CAMERA_SERVICE);
            Size bestSize = new Size(640,480);

            try {
                Size maxsize = new Size(640,480);
                for (String cameraId : cameraManager.getCameraIdList()) {
                    CameraCharacteristics characteristics
                            = cameraManager.getCameraCharacteristics(cameraId);

                    // We don't use a front facing camera in this sample.
                    Integer facing = characteristics.get(CameraCharacteristics.LENS_FACING);
                    if (facing != null && facing == CameraCharacteristics.LENS_FACING_FRONT) {
                        continue;
                    }
                    StreamConfigurationMap map = characteristics.get(CameraCharacteristics.SCALER_STREAM_CONFIGURATION_MAP);
                    Log.e("AK","configs "+map);
                    Size[] sizes = map.getOutputSizes(SurfaceTexture.class);
                    ArrayList<Size> correctSize =new ArrayList<>();
                    for (int i=0;i<sizes.length;i++)
                    {
                        Log.e("AK","sizes : "+sizes[i]);
                        float ar = (sizes[i].getWidth()*1.0f)/sizes[i].getHeight();
                        float ar169 = 16/9.0f;
                        float ar43 =4/3.0f;
                        float screenar = (screenAspectRatio==AspectRatio.RATIO_16_9)?ar169:ar43;
                        Log.e("AK","screenar"+screenar);
                        Log.e("AK","ar"+ar);
                        if ( ar == screenar)
                        {
                            if (sizes[i].getWidth()<=1920 && sizes[i].getHeight()<=1080)
                            {
                                correctSize.add(sizes[i]);
                                if (maxsize.getHeight()*maxsize.getWidth() <= sizes[i].getWidth()*sizes[i].getHeight())
                                {
                                    maxsize = sizes[i];
                                    bestSize = maxsize;
                                }
                                Log.e("AK","correctSize"+sizes[i]);
                            }

                        }

                    }
                }
            }
            catch (Exception e)
            {
                e.printStackTrace();
            }

owahltinez, 2020-01-20:
Hi @asifk1997, this repo is used to track issues with the sample. Since you are modifying the sample, I would suggest to open a question on StackOverflow (using tag `android-camerax`) or send an email to the [developer forums](g.co/camerax/developers).

closed 2020-01-20 by owahltinez