android/camera-samples #148: Old README Linking

Agnohendrix, 2020-01-18:
Report issue links in README at:

- [Camera2BasicSample](https://github.com/android/camera-samples/tree/master/Camera2BasicJava)

- [Camera2BasicKotlin](https://github.com/android/camera-samples/tree/master/Camera2BasicKotlin)

- [Camera2VideoSample](https://github.com/android/camera-samples/tree/master/Camera2VideoJava)

- [Camera2VideoKotlin](https://github.com/android/camera-samples/tree/master/Camera2VideoKotlin)

- [HDRViewFinder](https://github.com/android/camera-samples/tree/master/HdrViewfinder)

Links are rispectively directed to:

- [Camera2Basic](https://github.com/googlearchive/android-Camera2Basic)

- [Camera2Basic](https://github.com/googlearchive/android-Camera2Basic)

- [Camera2Video](https://github.com/googlearchive/android-Camera2Video)

- [Camera2Video](https://github.com/googlearchive/android-Camera2Video)

- [HDRViewFinder](https://github.com/googlearchive/android-HdrViewfinder)

Which are archived repository and should be replaced with rispective subfolders because they are migrated to [camera-samples](https://github.com/android/camera)

owahltinez, 2020-01-20:
The links work for me, what issue are you seeing?

syslogic, 2020-01-20:
@owahltinez It reads "This repository has been archived by the owner. It is now read-only." ...and so it might be difficult to file any issues. Also Google+ for consumers had been shutdown - and that community had been deleted along with everything else.

owahltinez, 2020-01-20:
Sorry, I think I'm missing which links are outdated. When I click on any of the links in the table from README they take me to the right place.

Agnohendrix, 2020-01-20:
If you look at the README.md in 

- https://github.com/android/camera-samples/tree/master/Camera2VideoJava

- https://github.com/android/camera-samples/tree/master/Camera2VideoKotlin

- https://github.com/android/camera-samples/tree/master/HdrViewfinder

They all have Issue report links that refer to https://github.com/googlearchive whilst (i guess) they should refer to their Issue page under https://github.com/android/camera-samples

owahltinez, 2020-01-21:
OK, I was only looking at the top-level README. Thanks for reporting!

syslogic, 2020-02-03:
@Agnohendrix The proper deep-link to Stackoverflow would be:
> http://stackoverflow.com/questions/tagged/android-camerax

And the referenced Google+ community does not exist anymore (it's a dead link):
> https://plus.google.com/communities/105153134372062985968

commit 8dbdf635 (2020-02-06): Changed README.md Issue links (android#148) (#161)

Fixes #148
closed 2020-02-06 by owahltinez