android/camera-samples #150: After close and open the screen two times, the App is not responding.

vw2k3, 2020-01-20:
Hello,
I'm trying Camera2VideoKotlin on the emulated Nexus 5X, API 28, Android 9.0 on Android studios. I've got an issue - the App is not responding after close and open the screen two times.

The following steps are to reproduce the issue:
1. Click [image] to close the screen.
2. Wait 2 seconds, and click button to open the screen. 
3. Repeat step 1 & 2, then the App is not responding like there's an residue shadow of camera covering the screen.

Any idea on how I can fix this issue?

owahltinez, 2020-01-21:
Can you confirm if this is happening on physical devices, or if it's an emulator only issue?

owahltinez, 2020-02-18:
Closing due to inactivity

closed 2020-02-18 by owahltinez

vw2k3, 2020-03-31:
Only on emulator ;)