android/camera-samples #151: CameraX 1.0.0-alpha06 some problems

dengly, 2020-01-20:
* CameraX 1.0.0-alpha06 can only open one camera, not multiple cameras at the same time
* CameraX 1.0.0-alpha06 has problem opening front camera, cannot preview

owahltinez, 2020-01-21:
> CameraX 1.0.0-alpha06 can only open one camera, not multiple cameras at the same time.

That is working by design, if you want multi-camera, you should use Camera2 API instead

> CameraX 1.0.0-alpha06 has problem opening front camera, cannot preview

A few questions that could help us narrow down the issue:
* Did you modify the sample code in any way?
* What is the exact device and model that you are experiencing this issue in?
* Can you provide clear reproduction steps?

owahltinez, 2020-01-28:
The sample has now been updated to `alpha08` via #154, please reopen if you are still experiencing any issues

closed 2020-01-28 by owahltinez