android/camera-samples #153: Video Capture issue - saved video flipped

mayurpateltristate, 2020-01-22:
library : androidx.camera:camera-core:1.0.0-alpha06
when i am recording video using above library from FRONT camera. Recorded video flipped
Please help me

owahltinez, 2020-01-23:
This repo is for issues about the sample. For issues related to usage of the library, please use the tag `android-camerax` in StackOverflow or join the [developer forums](g.co/camerax/developers).

closed 2020-01-23 by owahltinez

mayurpateltristate, 2020-01-28:
https://github.com/android/camera-samples/tree/master/Camera2VideoJava
we have used the above sample URL(Camera2VideoJava). but the same problem happens in the above sample also.
video flipped if recorded from the front camera.

owahltinez, 2020-01-28:
This is working as intended. The video is not flipped, the preview being shown to the user is. See this for more info: https://stackoverflow.com/a/18849598

mayurpateltristate, 2020-01-28:
But I want to record the same mirror video which I have seen in the preview. like Instagram, Snapchat, etc. How can I record video without flipped? On Instagram if you record from the front camera then you can see the video the same as preview. the recorded video should be the same as preview.

muzeitlimited, 2021-09-24:
Any fix for this? or a workaround? in case there is please inform me