android/camera-samples #155: CameraXBasic is Zoomed in

Ezugworie, 2020-01-22:
The Camera view is zoomed in automatically, please any idea on how to resolve this zooming to the normal view.

owahltinez, 2020-01-23:
Hello, can you please answer a few questions to help us better understand the issue?

* What device and model are you seeing this in?
* Are you making any modifications to the sample code?
* Can you please clarify what you mean by zoomed in? Pictures / screenshots pointing out the issue would greatly help

Ezugworie, 2020-01-23:
ok, thanks

- I am using this on a Techno Camon C12 (Andriod Version 9)

- Made a little modification to the naming convention of captured images and the 
side buttons

- On the Camera view, I get this:
[image]

while the captured image becomes this:
[image]

A close look at these images shows that the camera auto crops the images on both the top and bottom, I'm thinking its zooming the image

owahltinez, 2020-01-24:
This is expected. The reason is that `PreviewView` is performing center-cropping to fill the entire screen -- otherwise you would see black bars on the top and bottom, known as letter-boxing. When a picture is taken, it is not being cropped. If you want to match exactly what the preview sees, you will need to crop the resulting image yourself.

Ezugworie, 2020-01-24:
ok thanks, but please is there a way to zoom out the camera? so that at least I can see all the sides of the image captured

owahltinez, 2020-01-24:
The camera is not zoomed in. The view is being cropped because otherwise the camera would not fill the screen. To get exactly the same POV you can either:

1. Crop your image captured to match the screen size
2. Make the view that displays the preview smaller than the screen, matching the aspect ratio of the image capture

Ezugworie, 2020-01-27:
Thanks a lot, you have been very helpful

closed 2020-01-27 by Ezugworie