android/camera-samples #164: Crash while navigating back to preview from captured image

waliahimanshu, 2020-02-05:
Getting this error when running the sample.

Steps to reproduce.

1) Capture the photo.
2) Navigate to captured image on clicking Gallery icon.
3) Navigate back to camera preview and notice the crash

I am able to reproduce consistently 👍 

Device Name : Nokia 5.1
Model : TA-1061
Android version : 9

```
    Process: com.android.example.cameraxbasic, PID: 31287
    java.lang.IllegalStateException
        at androidx.core.util.Preconditions.checkState(Preconditions.java:96)
        at androidx.core.util.Preconditions.checkState(Preconditions.java:108)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.onClosed(Camera2CameraImpl.java:1298)
        at androidx.camera.camera2.internal.CameraDeviceStateCallbacks$ComboDeviceStateCallback.onClosed(CameraDeviceStateCallbacks.java:105)
        at android.hardware.camera2.impl.CameraDeviceImpl$5.run(CameraDeviceImpl.java:207)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)

```

owahltinez, 2020-02-05:
Hi Himanshu, can you please confirm if this is happening every time using your repro steps and that you are using the current version of the sample from the master branch?

I tried to reproduce the problem with the steps you provided in several devices, but I was unable to.

waliahimanshu, 2020-02-06:
Hi, just took the latest changes since yesterday, and got the crash using repro steps above but different stack trace (see below), this time.

However, yesterday I updated the version of sample project to alpha09, and the crash did not happen.
I also found this note in release notes of alpha 09
>Fixed issues when closing then opening cameras resulting in "Precondition" check failure.

So I reckon, the sample needs updating, because as is it crashes constantly.

```
2020-02-06 12:05:12.428 16038-16038/com.android.example.cameraxbasic D/AndroidRuntime: Shutting down VM
2020-02-06 12:05:12.429 16038-16079/com.android.example.cameraxbasic D/Camera: releasing session in state CLOSING
    
    --------- beginning of crash
2020-02-06 12:05:12.430 16038-16038/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.cameraxbasic, PID: 16038
    java.lang.NullPointerException: No camera bound to use case: ImageAnalysis:androidx.camera.core.ImageAnalysis-55973804-64fa-4cf8-b704-ddc45daef628
        at androidx.core.util.Preconditions.checkNotNull(Preconditions.java:81)
        at androidx.camera.core.UseCase.getBoundCameraId(UseCase.java:360)
        at androidx.camera.core.UseCase.isCurrentlyBoundCamera(UseCase.java:371)
        at androidx.camera.core.ImageAnalysis$1.onError(ImageAnalysis.java:212)
        at androidx.camera.camera2.internal.Camera2CameraImpl$16.run(Camera2CameraImpl.java:1008)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6715)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)```

YoungLeeForeverBoy, 2020-02-11:
hi, I got the same error when I run the CameraXBasic sample on my device:

Capture the photo. Nav to image preview, then back to camera preview screen.

Device Name : xiaomi 6
Android version : 9

```
Process: com.android.example.cameraxbasic, PID: 2060
    java.lang.IllegalStateException
        at androidx.core.util.Preconditions.checkState(Preconditions.java:96)
        at androidx.core.util.Preconditions.checkState(Preconditions.java:108)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.onClosed(Camera2CameraImpl.java:1298)
        at androidx.camera.camera2.internal.CameraDeviceStateCallbacks$ComboDeviceStateCallback.onClosed(CameraDeviceStateCallbacks.java:105)
        at android.hardware.camera2.impl.CameraDeviceImpl$5.run(CameraDeviceImpl.java:213)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:207)
        at android.os.HandlerThread.run(HandlerThread.java:65)
```

asifk1997, 2020-02-12:
Hi , i also got same issue even i upgraded to alpha09 .

Vivo 1951

Process: com.transpacks.CheckoH, PID: 15995
    java.lang.IllegalStateException
        at androidx.core.util.Preconditions.checkState(Preconditions.java:96)
        at androidx.core.util.Preconditions.checkState(Preconditions.java:108)
        at androidx.camera.view.TextureViewImplementation.lambda$null$0$TextureViewImplementation(TextureViewImplementation.java:78)
        at androidx.camera.view.-$$Lambda$TextureViewImplementation$TDSLJAQmFvVANkRF7SaUqnGsleY.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:224)
        at android.app.ActivityThread.main(ActivityThread.java:7055)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:536)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:876)

Vasniktel, 2020-02-12:
> Getting this error when running the sample.
>
> Steps to reproduce.
>
> 1. Capture the photo.
> 2. Navigate to captured image on clicking Gallery icon.
> 3. Navigate back to camera preview and notice the [...]

I'm getting the same error when I'm trying to switch between different cameras in CameraXBasic.
Device: Xiaomi redmi 4
Os: Android 7

P.S. Switching from preview to camera view also causes it

Simon-Richards, 2020-02-25:
I was getting this error also, and on screen rotation too. I did as @waliahimanshu suggested and upgraded to -alpha09 and the issue is gone

owahltinez, 2020-02-25:
We just changed back-navigation with #135 and I think it should fix this issue. Please reopen if you can still reproduce with the latest version of the sample.

closed 2020-02-25 by owahltinez

vanam, 2020-02-29:
It is still crashing.

Steps to reproduce.

1. Capture the photo. (Or skip this step if you already have done that.)
2. Navigate to captured image by clicking Gallery icon.
3. Navigate back to camera preview
4. **Take a picture and notice the crash.**

```
D/Camera: Use cases [Preview:androidx.camera.core.Preview-9b0aa555-4056-41a4-8142-29382f648059, ImageCapture:androidx.camera.core.ImageCapture-a9bb3213-f80f-404b-8f87-d1c7cba6b532, ImageAnalysis:androidx.camera.core.ImageAnalysis-87134d34-f7e5-4b40-8710-0ae308e5ead4] now ONLINE for camera 0
D/UseCaseAttachState: Active and online use case: [] for camera: 0
D/Camera: Resetting Capture Session
    releasing session in state CLOSING
    Transitioning camera internal state: CLOSING --> REOPENING
    Transitioning camera internal state: REOPENING --> OPENED
D/UseCaseAttachState: All use case: [androidx.camera.core.Preview-9b0aa555-4056-41a4-8142-29382f648059, androidx.camera.core.ImageCapture-a9bb3213-f80f-404b-8f87-d1c7cba6b532, androidx.camera.core.ImageAnalysis-87134d34-f7e5-4b40-8710-0ae308e5ead4] for camera: 0
D/Camera: Use case Preview:androidx.camera.core.Preview-9b0aa555-4056-41a4-8142-29382f648059 ACTIVE for camera 0
D/UseCaseAttachState: Active and online use case: [androidx.camera.core.Preview-9b0aa555-4056-41a4-8142-29382f648059] for camera: 0
D/Camera: Use case ImageCapture:androidx.camera.core.ImageCapture-a9bb3213-f80f-404b-8f87-d1c7cba6b532 ACTIVE for camera 0
D/UseCaseAttachState: Active and online use case: [androidx.camera.core.Preview-9b0aa555-4056-41a4-8142-29382f648059, androidx.camera.core.ImageCapture-a9bb3213-f80f-404b-8f87-d1c7cba6b532] for camera: 0
D/Camera: Use case ImageAnalysis:androidx.camera.core.ImageAnalysis-87134d34-f7e5-4b40-8710-0ae308e5ead4 ACTIVE for camera 0
D/UseCaseAttachState: Active and online use case: [androidx.camera.core.Preview-9b0aa555-4056-41a4-8142-29382f648059, androidx.camera.core.ImageCapture-a9bb3213-f80f-404b-8f87-d1c7cba6b532, androidx.camera.core.ImageAnalysis-87134d34-f7e5-4b40-8710-0ae308e5ead4] for camera: 0
D/CaptureSession: Opening capture session.
D/CaptureSession: Attempting to send capture request onConfigured
    Issuing request for session.
D/CaptureSession: CameraCaptureSession.onConfigured() mState=OPENED
D/CaptureSession: CameraCaptureSession.onReady() OPENED
    CameraCaptureSession.onReady() OPENED
D/CaptureSession: CameraCaptureSession.onClosed()
D/HostConnection: HostConnection::get() New Host Connection established 0xdf66f9c0, tid 21055
D/HostConnection: HostComposition ext ANDROID_EMU_CHECKSUM_HELPER_v1 ANDROID_EMU_dma_v1 ANDROID_EMU_YUV420_888_to_NV21 ANDROID_EMU_YUV_Cache ANDROID_EMU_async_unmap_buffer GL_OES_vertex_array_object GL_KHR_texture_compression_astc_ldr ANDROID_EMU_gles_max_version_2 
D/HostConnection: HostConnection::get() New Host Connection established 0xdf5ac280, tid 21087
D/HostConnection: HostComposition ext ANDROID_EMU_CHECKSUM_HELPER_v1 ANDROID_EMU_dma_v1 ANDROID_EMU_YUV420_888_to_NV21 ANDROID_EMU_YUV_Cache ANDROID_EMU_async_unmap_buffer GL_OES_vertex_array_object GL_KHR_texture_compression_astc_ldr ANDROID_EMU_gles_max_version_2 
D/ImageCapture: Size of image capture request queue: 0
D/Camera: issue capture request for camera 0
D/CaptureSession: Issuing capture request.
W/ExifInterface: Stop reading file since a wrong offset may cause an infinite loop: 0
    Skip the tag entry since data format (ULONG) is unexpected for tag: GPSAltitudeRef
    Stop reading file since a wrong offset may cause an infinite loop: 0
I/chatty: uid=10086(com.android.example.cameraxbasic) CameraX-camerax identical 1 line
W/ExifInterface: Stop reading file since a wrong offset may cause an infinite loop: 0
W/ExifInterface: Skip the tag entry since data format (ULONG) is unexpected for tag: GPSAltitudeRef
    Stop reading file since a wrong offset may cause an infinite loop: 0
E/AndroidRuntime: FATAL EXCEPTION: CameraX-camerax_io_0
    Process: com.android.example.cameraxbasic, PID: 21039
    java.util.concurrent.RejectedExecutionException: Task androidx.camera.core.-$$Lambda$ImageSaver$S9mrYGMPcUwPIjUa0oK9HMzbx_o@608e55f rejected from java.util.concurrent.ThreadPoolExecutor@9f29891[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 31]
        at java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2085)
        at java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:848)
        at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1394)
        at java.util.concurrent.Executors$DelegatedExecutorService.execute(Executors.java:624)
        at androidx.camera.core.ImageSaver.postSuccess(ImageSaver.java:240)
        at androidx.camera.core.ImageSaver.run(ImageSaver.java:182)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
W/ExifInterface: Stop reading file since a wrong offset may cause an infinite loop: 0
I/Process: Sending signal. PID: 21039 SIG: 9
```