android/camera-samples #168: Video recording with front camera in tablet(portrait mode) shows wrong aspect ratio.

PrekshyaBasnet, 2020-02-11:
Devices tested
Emulator: Pixel C API 29
Device: Galaxy Tab S2
Mode: Portrait

I tested the sample app for Kotlin version. And tried recording the video using `val cameraId = manager.cameraIdList[1] ` but the video aspect ratio doesn't seem to work properly.

[image]

owahltinez, 2020-02-15:
A few questions to help us find out the root cause of this issue:
* What sample are you seeing this in?
* Did you modify the sample in any way?

PrekshyaBasnet, 2020-02-16:
What sample are you seeing this in?
- It is in camera sample for Camera2VideoKotlin App.

Did you modify the sample in any way?
- I changed val cameraId = manager.cameraIdList[0] to val cameraId = manager.cameraIdList[1] for front camera.

commit 71e83b90 (2020-04-20): Fix layout issues

- Fixes #168
- Fixes #169
- Add capture button to all Camera2 samples
- Deprecate Camera2Video Java sample

Change-Id: I8cbdb7b01cd137b495d2e2b688399cd2077e7c2a
closed 2020-04-20 by owahltinez