android/camera-samples #169: Camera2Basic - AutoFitSurfaceView stretched and cropped with black space

rvhamed, 2020-02-12:
I have cropped issues on Samsung tablet SM-T331 (Android version: 5.1.1).
AutoFitSurfaceView stretched and cropped, as you see below, but when I tried to capture a photo, the result captured photo didn't get stretched and cropped anymore.
Note: When the application goes to background and returns to the foreground, this issue will be fixed.

Camera (SurfaceView):
[image]

Result Caputured photo (ImageReader):
[image]

owahltinez, 2020-02-15:
It seems like a potential race condition where the view is not appropriately measured during onCreate. Did you modify the sample in any way?

rvhamed, 2020-02-15:
@owahltinez 
No, it has not been changed.

owahltinez, 2020-02-18:
Can you confirm if this happens 100% of the time, or only sometimes? Is it only this specific device, or are you seeing this issue in other devices as well?

rvhamed, 2020-02-19:
I think this issue occurred on some specific devices. such as **Samsung tablet SM-T331 (Android version: 5.1.1).**
This is our scenario for the test:
1- Launch CameraFragment with background camera. (camera id 0) -> In this state the view was stretched.
2- Press the Home button. The application goes to the background
3- Open the Application from the recent list. -> In this state the view did not streched!! and issue fixed.

So this issue happens for the first time in each fragment's new launch.
We have tested it on Genymotion tablet size, it's not happened.

owahltinez, 2020-02-25:
Thank you. It looks like it may be related to #168 but I will need a device to be able to reproduce and confirm that.

Shuyun6, 2020-03-12:
I face the same problem! @owahltinez 
In Camera2Basic, project runs normally in Android 8 or higher. But in a Android 5.1(I have tested in 3 different phone) device, the preview shows the same as @rvhamed did.
Then I found that the problem may come from "surfaceHolder.setFixedSize()" method. Such as, when I choose the preview size 1920*1080, and use `holder.setFixedSize(1920, 1080)`, in Android 8 or higher, the preview is correct. But in Android 5.1, it should be `holder.setFixedSize(1080, 1920)` then the preview will be correct. 
So what is the different between Android 5.1 and Android 8(or higher) when I use `setFixedSize()` ? And how can I detect When to use `holder.setFixedSize(1920, 1080)` or `holder.setFixedSize(1080, 1920)`?

Shuyun6, 2020-03-12:
According to @rvhamed 's test scenario and my view in `setFixedSize` in previous comments, I have solved this problem by calling `setFixedSize` before `surfaceCreated`. Actually, I put `view.holder.setFixedSize` in front of `view.holder.addCallback` in Camera2Basic project.
So here is my new problem, what is the best place to call `setFixedSize()` ?

rvhamed, 2020-03-14:
@Shuyun6 Thanks, It works. I tested it on Android 5.1 and Android 9.
     
        val previewSize = getPreviewOutputSize(viewFinder.display, characteristics, SurfaceHolder::class.java)
        view.holder.setFixedSize(previewSize.width, previewSize.height)

        view.holder.addCallback(object : SurfaceHolder.Callback {
            override fun surfaceDestroyed(holder: SurfaceHolder) = Unit

            override fun surfaceChanged(
                    holder: SurfaceHolder,
                    format: Int,
                    width: Int,
                    height: Int) = Unit

            override fun surfaceCreated(holder: SurfaceHolder) {

                // Selects appropriate preview size and configures view finder
                Log.d(TAG, "View finder size: ${viewFinder.width} x ${viewFinder.height}")
                Log.d(TAG, "Selected preview size: $previewSize")
                view.setAspectRatio(previewSize.width, previewSize.height)

                // To ensure that size is set, initialize camera in the view's thread
                view.post { initializeCamera() }
            }
        })

> So here is my new problem, what is the best place to call setFixedSize() ?

And this is my question too.

rvhamed, 2020-03-14:
According to @Shuyun6 solution, I found a new solution. When I moved `setFixedSize()` method to above of `addCallBack()` method in our project(Not SampleProject), I got NullPointerException for  viewFinder.display that couldn't be null.
In the new solution, I just called `setSizeFromLayout()` method within `surfaceCreated()`.
This worked fine in Samsung tablet SM-T331 (Android version: 5.1.1). But this causes a new problem in Samsung J7 (Android version: 8.1), The view stretched.

      view.holder.addCallback(object : SurfaceHolder.Callback {
            override fun surfaceDestroyed(holder: SurfaceHolder) = Unit

            override fun surfaceChanged(
                holder: SurfaceHolder,
                format: Int,
                width: Int,
                height: Int
            ) = Unit

            override fun surfaceCreated(holder: SurfaceHolder) {

                // Selects appropriate preview size and configures view finder
                val previewSize = getPreviewOutputSize(
                    viewFinder.display, characteristics, SurfaceHolder::class.java
                )
                Log.d(TAG, "View finder size: ${viewFinder.width} x ${viewFinder.height}")
                Log.d(TAG, "Selected preview size: $previewSize")
                view.holder.setFixedSize(previewSize.width, previewSize.height)
               
                view.holder.setSizeFromLayout()
                
                view.setAspectRatio(previewSize.width, previewSize.height)

                // Check if the flash is supported.
                flashSupported =
                    characteristics.get(CameraCharacteristics.FLASH_INFO_AVAILABLE) == true

                // To ensure that size is set, initialize camera in the view's thread
                view.post { initializeCamera() }
            }
        })

Shuyun6, 2020-03-14:
@rvhamed It's not necessary to use `setSizeFromLayout` after `setFixedSize`,  it will resize the surface's size, cause the size need to be the preview size, but not the layout size

commit 71e83b90 (2020-04-20): Fix layout issues

- Fixes #168
- Fixes #169
- Add capture button to all Camera2 samples
- Deprecate Camera2Video Java sample

Change-Id: I8cbdb7b01cd137b495d2e2b688399cd2077e7c2a
closed 2020-04-20 by owahltinez