android/camera-samples #172: No static field fragment_container

Ezugworie, 2020-02-19:
I am getting this error suddenly 
here is the log

```java.lang.NoSuchFieldError: No static field fragment_contain of type I in class Lcom/android/example/cameraxbasic/R$id; or its superclasses (declaration of 'com.android.example.cameraxbasic.R$id' appears in /data/app/com.virtusync.scanningtool-BeCKdrMVet4imgvHfjLaeg==/base.apk!classes2.dex)
        at com.android.example.cameraxbasic.MainWarrantActivity.onCreate(MainWarrantActivity.kt:52)```

syslogic, 2020-02-24:
There is no `MainWarrantActivity.kt` and that probably should be `R.id.fragment_container`.

owahltinez, 2020-02-25:
@Ezugworie I'm guessing this is for the `CameraXBasic` sample, correct? Are you modifying the sample in any way? Can you please provide reproduction steps for this error?

owahltinez, 2020-03-30:
Closing due to inactivity.

closed 2020-03-30 by owahltinez