android/camera-samples #177: The photo rotation

Winny-Wiky, 2020-02-28:
Hello，I meet some bug in CameraX(Alpha-10)，please help me.
When the photo is captured. I find the photo rotation is different in different phone, example, the photo has rotated 90 in OnePlus 5T(Android 9.1)，the photo is correct in MI 5（Android 8.1）

owahltinez, 2020-03-04:
Thanks for reporting this, it looks like you are not using the latest version of `CameraXBasic` which is now using `beta01`. Please try again with the latest sample, without any modifications, and reopen if you are still seeing an issue.

closed 2020-03-04 by owahltinez

Winny-Wiky, 2020-03-05:
Hi, this issue still exists in the latest version(beta01). I used the function that is
`public void takePicture(@NonNull Executor executor,final @NonNull OnImageCapturedCallback callback)`，I created a bitmap with the image byteArray. Maybe there is something wrong with this place ? Thanks.