android/camera-samples #178: Using FragmentContainerView in main_activity.xml

syslogic, 2020-02-29:
[`androidx.fragment.app.FragmentContainerView`](https://developer.android.com/reference/kotlin/androidx/fragment/app/FragmentContainerView) seems to replace `FrameLayout` as the fragment host:

> `FragmentContainerView` is a customized Layout designed specifically for Fragments. It extends `FrameLayout`, so it can reliably handle Fragment Transactions, and it also has additional features to coordinate with fragment behavior.

owahltinez, 2020-03-04:
This makes sense, although I am a bit wary of adding dependencies for the sake of following guidance alone. What issues are we seeing that would warrant using `FragmentContainerView`?

owahltinez, 2020-03-30:
It looks like this produces a lint error, so it would make sense to move to `FragmentContainerView` considering how important that appears to be to Android Studio.

owahltinez, 2020-05-26:
CameraX is already using FragmentContainerView!

closed 2020-05-26 by owahltinez