android/camera-samples #179: Black preview in CameraXBasic

zsoerenm, 2020-02-29:
Hi,
I have installed CameraXBasic without any modification on my phone (HTC One). When I open the app, the preview is black and stays black whereas it should preview the back camera. The preview for the front camera is okay and I can capture photos with the front camera.
Here is a logcat after opening the app:
```
2020-02-29 19:45:09.280 29062-29062/? I/le.cameraxbasi: Late-enabling -Xcheck:jni
2020-02-29 19:45:09.337 29062-29062/? E/le.cameraxbasi: Unknown bits set in runtime_flags: 0x8000
2020-02-29 19:45:10.532 29062-29062/com.android.example.cameraxbasic W/le.cameraxbasi: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (greylist, reflection, allowed)
2020-02-29 19:45:10.534 29062-29062/com.android.example.cameraxbasic W/le.cameraxbasi: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (greylist, reflection, allowed)
2020-02-29 19:45:11.014 29062-29116/com.android.example.cameraxbasic I/Adreno-EGL: <qeglDrvAPI_eglInitialize:379>: QUALCOMM Build: 10/21/15, 369a2ea, I96aee987eb
2020-02-29 19:45:11.028 29062-29116/com.android.example.cameraxbasic W/Adreno-EGL: <qeglDrvAPI_eglGetConfigAttrib:607>: EGL_BAD_ATTRIBUTE
2020-02-29 19:45:11.043 29062-29116/com.android.example.cameraxbasic W/Adreno-EGL: <qeglDrvAPI_eglGetConfigAttrib:607>: EGL_BAD_ATTRIBUTE
2020-02-29 19:45:11.078 29062-29116/com.android.example.cameraxbasic W/Gralloc3: mapper 3.x is not supported
2020-02-29 19:45:11.304 29062-29062/com.android.example.cameraxbasic D/CameraXBasic: Screen metrics: 1080 x 1920
2020-02-29 19:45:11.304 29062-29062/com.android.example.cameraxbasic D/CameraXBasic: Preview aspect ratio: 1
2020-02-29 19:45:11.376 29062-29131/com.android.example.cameraxbasic I/CameraManagerGlobal: Connecting to camera service
2020-02-29 19:45:11.414 29062-29062/com.android.example.cameraxbasic W/Glide: Failed to find GeneratedAppGlideModule. You should include an annotationProcessor compile dependency on com.github.bumptech.glide:compiler in your application and a @GlideModule annotated AppGlideModule implementation or LibraryGlideModules will be silently ignored
2020-02-29 19:45:11.523 29062-29131/com.android.example.cameraxbasic D/CameraRepository: Added camera: 0
2020-02-29 19:45:11.607 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [] for camera: 0
2020-02-29 19:45:11.609 29062-29131/com.android.example.cameraxbasic I/Camera2CameraInfo: Device Level: INFO_SUPPORTED_HARDWARE_LEVEL_LEGACY
2020-02-29 19:45:11.648 29062-29131/com.android.example.cameraxbasic D/CameraRepository: Added camera: 1
2020-02-29 19:45:11.712 29062-29131/com.android.example.cameraxbasic I/Camera2CameraInfo: Device Level: INFO_SUPPORTED_HARDWARE_LEVEL_LEGACY
2020-02-29 19:45:11.714 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [] for camera: 1
2020-02-29 19:45:12.085 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [] for camera: 0
2020-02-29 19:45:12.088 29062-29132/com.android.example.cameraxbasic D/Camera: Use cases [Preview:androidx.camera.core.Preview-18e87669-99cf-4b7f-aff9-05e15d0cf2e3, ImageCapture:androidx.camera.core.ImageCapture-d43b50a7-e06e-4c43-ac26-5f4876401b07, ImageAnalysis:androidx.camera.core.ImageAnalysis-a98eabb0-f766-464d-8aa9-e02e7ae0d6ad] now ONLINE for camera 0
2020-02-29 19:45:12.088 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [] for camera: 0
2020-02-29 19:45:12.088 29062-29132/com.android.example.cameraxbasic D/Camera: Resetting Capture Session
2020-02-29 19:45:12.089 29062-29132/com.android.example.cameraxbasic D/Camera: releasing session in state INITIALIZED
2020-02-29 19:45:12.090 29062-29132/com.android.example.cameraxbasic D/Camera: Transitioning camera internal state: INITIALIZED --> OPENING
2020-02-29 19:45:12.090 29062-29132/com.android.example.cameraxbasic D/Camera: Opening camera: 0
2020-02-29 19:45:12.092 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: All use case: [androidx.camera.core.Preview-18e87669-99cf-4b7f-aff9-05e15d0cf2e3, androidx.camera.core.ImageCapture-d43b50a7-e06e-4c43-ac26-5f4876401b07, androidx.camera.core.ImageAnalysis-a98eabb0-f766-464d-8aa9-e02e7ae0d6ad] for camera: 0
2020-02-29 19:45:12.126 29062-29132/com.android.example.cameraxbasic I/CameraManager: Using legacy camera HAL.
2020-02-29 19:45:12.692 29062-29132/com.android.example.cameraxbasic D/Camera: Use case Preview:androidx.camera.core.Preview-18e87669-99cf-4b7f-aff9-05e15d0cf2e3 ACTIVE for camera 0
2020-02-29 19:45:12.693 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [androidx.camera.core.Preview-18e87669-99cf-4b7f-aff9-05e15d0cf2e3] for camera: 0
2020-02-29 19:45:12.694 29062-29132/com.android.example.cameraxbasic D/Camera: Use case ImageCapture:androidx.camera.core.ImageCapture-d43b50a7-e06e-4c43-ac26-5f4876401b07 ACTIVE for camera 0
2020-02-29 19:45:12.696 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [androidx.camera.core.Preview-18e87669-99cf-4b7f-aff9-05e15d0cf2e3, androidx.camera.core.ImageCapture-d43b50a7-e06e-4c43-ac26-5f4876401b07] for camera: 0
2020-02-29 19:45:12.697 29062-29132/com.android.example.cameraxbasic D/Camera: Use case ImageAnalysis:androidx.camera.core.ImageAnalysis-a98eabb0-f766-464d-8aa9-e02e7ae0d6ad ACTIVE for camera 0
2020-02-29 19:45:12.698 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [androidx.camera.core.Preview-18e87669-99cf-4b7f-aff9-05e15d0cf2e3, androidx.camera.core.ImageCapture-d43b50a7-e06e-4c43-ac26-5f4876401b07, androidx.camera.core.ImageAnalysis-a98eabb0-f766-464d-8aa9-e02e7ae0d6ad] for camera: 0
2020-02-29 19:45:12.700 29062-29132/com.android.example.cameraxbasic D/Camera: CameraDevice.onOpened(): 0
2020-02-29 19:45:12.707 29062-29132/com.android.example.cameraxbasic D/Camera: Transitioning camera internal state: OPENING --> OPENED
2020-02-29 19:45:12.709 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: All use case: [androidx.camera.core.Preview-18e87669-99cf-4b7f-aff9-05e15d0cf2e3, androidx.camera.core.ImageCapture-d43b50a7-e06e-4c43-ac26-5f4876401b07, androidx.camera.core.ImageAnalysis-a98eabb0-f766-464d-8aa9-e02e7ae0d6ad] for camera: 0
2020-02-29 19:45:12.719 29062-29132/com.android.example.cameraxbasic D/CaptureSession: Opening capture session.
2020-02-29 19:45:12.747 29062-29132/com.android.example.cameraxbasic I/CameraDeviceState: Legacy camera service transitioning to state CONFIGURING
2020-02-29 19:45:12.750 29062-29167/com.android.example.cameraxbasic I/RequestThread-0: Configure outputs: 3 surfaces configured.
2020-02-29 19:45:12.750 29062-29167/com.android.example.cameraxbasic D/Camera: app passed NULL surface
2020-02-29 19:45:12.759 29062-29167/com.android.example.cameraxbasic I/RequestThread-0: configureOutputs - set take picture size to 2688x1520
2020-02-29 19:45:12.771 29062-29168/com.android.example.cameraxbasic W/Adreno-EGL: <qeglDrvAPI_eglGetConfigAttrib:607>: EGL_BAD_ATTRIBUTE
2020-02-29 19:45:12.773 29062-29168/com.android.example.cameraxbasic W/Adreno-EGL: <qeglDrvAPI_eglGetConfigAttrib:607>: EGL_BAD_ATTRIBUTE
2020-02-29 19:45:12.794 29062-29168/com.android.example.cameraxbasic W/Adreno-EGL: <qeglDrvAPI_eglQueryContext:4373>: EGL_BAD_ATTRIBUTE
2020-02-29 19:45:12.795 29062-29168/com.android.example.cameraxbasic E/libc: Access denied finding property "persist.camera.legacy_perf"
2020-02-29 19:45:12.814 29062-29132/com.android.example.cameraxbasic I/CameraDeviceState: Legacy camera service transitioning to state IDLE
2020-02-29 19:45:12.819 29062-29132/com.android.example.cameraxbasic D/CaptureSession: Attempting to send capture request onConfigured
2020-02-29 19:45:12.819 29062-29132/com.android.example.cameraxbasic D/CaptureSession: Issuing request for session.
2020-02-29 19:45:12.835 29062-29132/com.android.example.cameraxbasic I/RequestQueue: Repeating capture request set.
2020-02-29 19:45:12.836 29062-29132/com.android.example.cameraxbasic D/CaptureSession: CameraCaptureSession.onConfigured() mState=OPENED
2020-02-29 19:45:12.836 29062-29132/com.android.example.cameraxbasic D/CaptureSession: CameraCaptureSession.onReady() OPENED
[... cut ...]
2020-02-29 19:45:36.545 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [] for camera: 0
2020-02-29 19:45:36.550 29062-29132/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [] for camera: 0
2020-02-29 19:45:36.707 29062-29193/com.android.example.cameraxbasic I/CameraManagerGlobal: Connecting to camera service
2020-02-29 19:45:36.707 29062-29193/com.android.example.cameraxbasic E/CameraManagerGlobal: Camera service is unavailable
```

owahltinez, 2020-03-04:
Thanks for reporting this! Can you confirm if the camera app that came preinstalled with your device displays the preview correctly? How about other third party apps?

zsoerenm, 2020-03-05:
I have custom ROM installed (LineageOS). The preinstalled camera is Footej Camera 2.4.10. The preview and capturing always work with this app. Other apps like for example Lightening QR (to scan a QR code), Google Duo (for video conference) etc. also work fine from the first instance.

owahltinez, 2020-03-05:
Unfortunately, we can't guarantee support for custom ROMs since they may not pass CTS (Android's compatibility testing suite). Please reopen if you can reproduce this issue on a certified Android device; you could try installing a [Generic System Image](https://source.android.com/setup/build/gsi) for example.

closed 2020-03-05 by owahltinez