android/camera-samples #181: method "stopRecordingVideo"  exception

salmanit, 2020-03-02:
device: android 6.0.1  samsung  SM-T350
click "Record" button is   normal , but after click "stop" button,  app will quit about few seconds.
we need repeat record videos,but now stopRecord will occur an error.
log:
`03-02 16:01:47.153 9320-9320/com.example.android.camera2video D/ViewRootImpl: ViewPostImeInputStage processPointer 0
03-02 16:01:47.243 9320-9320/com.example.android.camera2video D/ViewRootImpl: ViewPostImeInputStage processPointer 1
03-02 16:01:47.293 9320-9454/com.example.android.camera2video W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
03-02 16:01:47.293 9320-9454/com.example.android.camera2video W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3800>: EGL_BAD_SURFACE
03-02 16:01:47.293 9320-9454/com.example.android.camera2video E/CameraDeviceGLThread-1: Received exception on GL render thread: 
    java.lang.IllegalStateException: swapBuffers: EGL error: 0x300d
        at android.hardware.camera2.legacy.SurfaceTextureRenderer.checkEglError(SurfaceTextureRenderer.java:530)
        at android.hardware.camera2.legacy.SurfaceTextureRenderer.swapBuffers(SurfaceTextureRenderer.java:523)
        at android.hardware.camera2.legacy.SurfaceTextureRenderer.drawIntoSurfaces(SurfaceTextureRenderer.java:727)
        at android.hardware.camera2.legacy.GLThreadManager$1.handleMessage(GLThreadManager.java:105)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:148)
        at android.os.HandlerThread.run(HandlerThread.java:61)
03-02 16:01:47.293 9320-9454/com.example.android.camera2video I/CameraDeviceState: Legacy camera service transitioning to state ERROR
03-02 16:01:47.343 9320-9320/com.example.android.camera2video D/Camera2VideoFragment: Video saved: /storage/emulated/0/1583136100848.mp4
03-02 16:01:47.353 9320-9320/com.example.android.camera2video E/CameraCaptureSession: Session 1: Exception while stopping repeating: 
    android.hardware.camera2.CameraAccessException: The camera device has encountered a serious error
        at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2000)
        at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:913)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:378)
        at com.example.android.camera2video.Camera2VideoFragment.closePreviewSession(Camera2VideoFragment.java:674)
        at com.example.android.camera2video.Camera2VideoFragment.startPreview(Camera2VideoFragment.java:496)
        at com.example.android.camera2video.Camera2VideoFragment.stopRecordingVideo(Camera2VideoFragment.java:694)
        at com.example.android.camera2video.Camera2VideoFragment.onClick(Camera2VideoFragment.java:314)
        at android.view.View.performClick(View.java:5714)
        at android.widget.TextView.performClick(TextView.java:10926)
        at android.view.View$PerformClick.run(View.java:22589)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.app.ActivityThread.main(ActivityThread.java:7325)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1321)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1211)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err: android.hardware.camera2.CameraAccessException: The camera device has encountered a serious error
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2000)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureRequest(CameraDeviceImpl.java:623)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at com.example.android.camera2video.Camera2VideoFragment.startPreview(Camera2VideoFragment.java:500)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at com.example.android.camera2video.Camera2VideoFragment.stopRecordingVideo(Camera2VideoFragment.java:694)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at com.example.android.camera2video.Camera2VideoFragment.onClick(Camera2VideoFragment.java:314)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at android.view.View.performClick(View.java:5714)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at android.widget.TextView.performClick(TextView.java:10926)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at android.view.View$PerformClick.run(View.java:22589)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at android.os.Looper.loop(Looper.java:148)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:7325)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1321)
03-02 16:01:47.353 9320-9320/com.example.android.camera2video W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1211)
03-02 16:01:51.263 9320-9438/com.example.android.camera2video E/RequestThread-1: Timed out while waiting for request to complete.
03-02 16:01:51.263 9320-9438/com.example.android.camera2video W/CaptureCollector: Preview buffers dropped for request: 1
03-02 16:01:51.263 9320-9438/com.example.android.camera2video E/CameraDeviceState: Cannot receive result while in state: 0
03-02 16:01:51.263 9320-9438/com.example.android.camera2video E/CameraDeviceState: Cannot receive result while in state: 0
03-02 16:01:51.263 9320-9438/com.example.android.camera2video E/CameraDeviceState: Cannot receive result while in state: 0
03-02 16:01:51.283 9320-9496/com.example.android.camera2video W/MessageQueue: Handler (android.graphics.SurfaceTexture$1) {6c08eb6} sending message to a Handler on a dead thread
    java.lang.IllegalStateException: Handler (android.graphics.SurfaceTexture$1) {6c08eb6} sending message to a Handler on a dead thread
        at android.os.MessageQueue.enqueueMessage(MessageQueue.java:543)
        at android.os.Handler.enqueueMessage(Handler.java:631)
        at android.os.Handler.sendMessageAtTime(Handler.java:600)
        at android.os.Handler.sendMessageDelayed(Handler.java:570)
        at android.os.Handler.sendEmptyMessageDelayed(Handler.java:534)
        at android.os.Handler.sendEmptyMessage(Handler.java:519)
        at android.graphics.SurfaceTexture.postEventFromNative(SurfaceTexture.java:376)
03-02 16:01:51.323 9320-9332/com.example.android.camera2video E/BufferQueueProducer: [SurfaceTexture-1-9320-2] queueBuffer: BufferQueue has been abandoned
03-02 16:01:51.323 9320-9496/com.example.android.camera2video E/BufferQueueProducer: [SurfaceTexture-1-9320-2] dequeueBuffer: BufferQueue has been abandoned
03-02 16:01:51.363 9320-9331/com.example.android.camera2video E/BufferQueueProducer: [SurfaceTexture-1-9320-2] cancelBuffer: BufferQueue has been abandoned
03-02 16:01:51.363 9320-9332/com.example.android.camera2video E/BufferQueueProducer: [SurfaceTexture-1-9320-2] cancelBuffer: BufferQueue has been abandoned
03-02 16:01:51.363 9320-9496/com.example.android.camera2video E/BufferQueueProducer: [SurfaceTexture-1-9320-2] cancelBuffer: BufferQueue has been abandoned
03-02 16:01:51.363 9320-9331/com.example.android.camera2video E/BufferQueueProducer: [SurfaceTexture-1-9320-2] cancelBuffer: BufferQueue has been abandoned
03-02 16:01:51.363 9320-9332/com.example.android.camera2video E/BufferQueueProducer: [SurfaceTexture-1-9320-2] cancelBuffer: BufferQueue has been abandoned
03-02 16:01:51.363 9320-9496/com.example.android.camera2video E/BufferQueueProducer: [SurfaceTexture-1-9320-2] cancelBuffer: BufferQueue has been abandoned`

owahltinez, 2020-03-30:
@salmanit thanks for reporting this and the detailed description of the issue. Sorry for the delay, I'll take a look at this and hopefully get back to you by the end of the week. We plan on refactoring the Camera2Video sample to make it more like the Camera2SlowMotion one, so potentially that will take care of this problem.

owahltinez, 2020-04-20:
The video sample was updated, which should fix this issue since SurfaceTexture is no longer being used

closed 2020-04-20 by owahltinez