android/camera-samples #185: the photo rotation

Winny-Wiky, 2020-03-05:
Hi, this issue still exists in the latest version(beta01). I used the function that is
public void takePicture(@NonNull Executor executor,final @NonNull OnImageCapturedCallback callback)，I created a bitmap with the image byteArray. Maybe there is something wrong with this place ? Thanks.

Winny-Wiky, 2020-03-05:
I used the CameraXBasic Demo(beta01) capture a picture, it show correct in photos app, but rotate -90 degree in my app, this is the code `   val cursor = contentResolver.query(
            MediaStore.Images.Media.EXTERNAL_CONTENT_URI,
            null,
            null,
            null,
            null
        )
        if (cursor != null && cursor.moveToLast()) {
            val data = cursor.getBlob(cursor.getColumnIndex(MediaStore.Images.Media.DATA))
            val filePath = String(data, 0, data.size - 1)
            photo_circle.setImageBitmap(BitmapFactory.decodeFile(filePath))
            cursor.close()
        }` , can you help me solve it, thanks.

owahltinez, 2020-03-05:
The pixels of the image are not rotated, it is using EXIF metadata which must be read by your application in order to properly rotate the image upon reading it. Libraries like Glide do this automatically.

closed 2020-03-05 by owahltinez