android/camera-samples #189: Runtime exception on CameraFragment$setGalleryThumbnail$1.run(CameraFragment.kt:178)

PawKanarek, 2020-03-10:
Hi i have run sample this sample project without any changes on my Huawei Mate 10 Pro (Android 9), then i've captured some photos, went into gallery, removed 1 photo, came back to camera fragment, then i've rotated device mulitple times to see how smooth rotation is. And after next few rotations app crashes with this on logcat: 

I've compiled codes from master (10.03.2020)

```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.cameraxbasic, PID: 24229
    java.lang.IllegalStateException: Fragment CameraFragment{86d08e2} (eedd5652-ee06-4fd1-b61b-30c8b3cb5aa4)} not attached to a context.
        at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
        at androidx.fragment.app.Fragment.getResources(Fragment.java:869)
        at com.android.example.cameraxbasic.fragments.CameraFragment$setGalleryThumbnail$1.run(CameraFragment.kt:178)
        at android.os.Handler.handleCallback(Handler.java:907)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:216)
        at android.app.ActivityThread.main(ActivityThread.java:7625)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)
```

owahltinez, 2020-03-30:
@PawKanarek thanks for reporting this and the detailed description of the issue. Sorry for the delay, I'll take a look at this and hopefully get back to you by the end of the week.

owahltinez, 2020-04-27:
Actually, it looks like the issue is completely unrelated to the CameraX library -- it's just a fragment timing issue. I'm closing it as a duplicate of #178 which should fix this problem.

closed 2020-04-27 by owahltinez