android/camera-samples #196: Flash issue with Camera2

buzmaxx, 2020-04-03:
CaptureRequest.CONTROL_AE_MODE_ON_ALWAYS_FLASH and CaptureRequest.CONTROL_AE_MODE_ON_AUTO_FLASH don't affect flash. Flash only fires, if CaptureRequest.FLASH_MODE set to FLASH_MODE_SINGLE or FLASH_MODE_TORCH, but in this case exposure works wrong. 

capture request is TEMPLATE_STILL_CAPTURE

set up is the same with you sample(camera2basics) with only diff - I added to request builder values from above, per documentation it should work

owahltinez, 2020-04-06:
Are you seeing this issue in all devices, or only a particular device? What devices have you tested? Please include specific model and Android version.

owahltinez, 2020-05-19:
Closing due to inactivity

closed 2020-05-19 by owahltinez

lawr1292, 2023-11-03:
I am having a similar issue. I replicated the camera2basic sample, and for my use case I need flash to fire every time. In the setAutoFlash() method I switched requestBuilder.set(CaptureRequest.CONTROL_AE_MODE, CaptureRequest.CONTROL_AE_MODE_ON_AUTO_FLASH); to CONTROL_AE_MODE_ON_ALWAYS_FLASH. The flash fires correctly everytime but the images come out way over-exposed, practically white. 

I am testing on a Google Pixel 6.