android/camera-samples #197: Firebase Face Detection Api issue while using cameraX API

Anasmirza20, 2020-04-06:
Hi team,

I used Camera X Api for detecting faces in Firebase ML kit because you used the depreciated camera API in our example and I faced a lot of performance issues in it.
CameraX API works perfectly but I am stuck in one place. CameraX uses Texture view in the layout view to open the camera and I want to show 3d objects and filters on the face.
In the example, we use CameraSourcePreview and GraphicOverlay classes in the quickstart sample app to show objects on the relative part of the face.
So, Could you please let me know how to how to achieve the same thing using cameraX API.

Any help will be appreciated.
Thanks

owahltinez, 2020-04-06:
Hello @Anasmirza20, we will be adding a sample to this repo which uses CameraX + MLKit -- please stay tuned!

Anasmirza20, 2020-04-06:
Okay, thanks but could you please let me know how much time you will take??

owahltinez, 2020-04-06:
In the best case scenario I won't be able to have a sample ready for at least another 2-4 weeks. The work was originally planned for early May.

Anasmirza20, 2020-04-06:
okay, thanks for instant response,
I hope you guys will do this asap.
Keep it up....

vitrol4, 2020-05-19:
@owahltinez any news?

Thivarella, 2020-05-19:
@owahltinez do you have any news when it's going to be ready?

owahltinez, 2020-06-26:
The MLKit repository of samples has now been updated to use the latest version of CameraX: https://github.com/googlesamples/mlkit

closed 2020-06-26 by owahltinez