android/camera-samples #200: Have problems with memory usage in Camera2BasicKotlin, same as #73

liungkejin, 2020-04-10:
https://github.com/android/camera-samples/issues/73#issue-500121477

native memory is unstable and always growing up, even until to OOM, test devices: Nokia x7/9.0

owahltinez, 2020-04-10:
Camera2BasicKotlin was deprecated, please try with Camera2Basic and reopen if you can still reproduce the issue.

closed 2020-04-10 by owahltinez

liungkejin, 2020-04-12:
I tried many solutions, the only solution is forcing GC, but not call `System.gc()`
```java
    private byte [] fuckingData = null;
    private void fixFuckingCam2OOM() {
        if (fuckingData == null) {
            fuckingData = new byte[1024 * 1024];
        } else {
            fuckingData = null;
        }
    }
```
Just call this method every camera frame! this will cause system gc per second!
My devices is `Nokia x7 / 9.0`

owahltinez, 2020-04-13:
I ran the sample for over 30 minutes on an emulator with a profiler attached to it. Memory slowly creeped up at first, but eventually became stable. Please make sure that you are using the latest version of the sample.

[image]

liungkejin, 2020-04-16:
> I ran the sample for over 30 minutes on an emulator with a profiler attached to it. Memory slowly creeped up at first, but eventually became stable. Please make sure that you are using the latest version of the sample.
> 
> [image]

I run the newest sample with newest android studio 3.6.2 on Nokia x7/9.0, here is my memory profiler record
[image]

owahltinez, 2020-04-23:
I see that the GC is kicking in at semi-regular intervals. Either the memory keeps growing until it reaches OOO, in which case it's an issue with the OEM (in this case Nokia) or it will eventually taper off with the GC keeping it in check.