android/camera-samples #203: A resource was acquired at attached stack trace but never released - kotlin

knightfork, 2020-04-14:
I am using camera2 and getting the following exception:

```
04-13 20:39:54.801 13097-13110/com.x.y.z E/StrictMode: A resource was acquired at attached stack trace but never released. See java.io.Closeable for information on avoiding resource leaks.
    java.lang.Throwable: Explicit termination method 'release' not called
        at dalvik.system.CloseGuard.open(CloseGuard.java:184)
        at android.view.Surface.setNativeObjectLocked(Surface.java:460)
        at android.view.Surface.<init>(Surface.java:152)
        at android.media.ImageReader.nativeGetSurface(Native Method)
        at android.media.ImageReader.<init>(ImageReader.java:130)
        at android.media.ImageReader.newInstance(ImageReader.java:100)
        at androidx.camera.core.MetadataImageReader.createImageReaderProxy(MetadataImageReader.java:124)
        at androidx.camera.core.MetadataImageReader.<init>(MetadataImageReader.java:113)
        at androidx.camera.core.ImageCapture.createPipeline(ImageCapture.java:335)
        at androidx.camera.core.ImageCapture.onSuggestedResolutionUpdated(ImageCapture.java:974)
        at androidx.camera.core.UseCase.updateSuggestedResolution(UseCase.java:372)
        at androidx.camera.core.CameraX.bindToLifecycle(CameraX.java:321)
        at androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:229)
        at com.x.y.z.fragments.CameraFragment$bindCameraUseCases$1.run(CameraFragment.kt:286)
        at android.os.Handler.handleCallback(Handler.java:815)
        at android.os.Handler.dispatchMessage(Handler.java:104)
        at android.os.Looper.loop(Looper.java:194)
        at android.app.ActivityThread.main(ActivityThread.java:5651)
        at java.lang.reflect.Method.invoke(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:372)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:959)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:754)
```

Looking at the stack trace suggests it's coming from SurfaceView not having been released. Checking reference suggests that while in Java you have to release it; not quite sure how to deal with it in Kotlin.

Btw., I have modified CameraXBasic very slightly so that on touching the gallery button it runs the following code:

```
    // Listener for button used to view the most recent photo
    camControls.findViewById<ImageButton>(R.id.photo_view_button).setOnClickListener {
        pickFromGallery()
    }

private fun pickFromGallery() {
    val intent = Intent(Intent.ACTION_PICK)
    intent.type = "image/*"
    val mimeTypes =
        arrayOf("image/jpeg", "image/png")
    intent.putExtra(Intent.EXTRA_MIME_TYPES, mimeTypes)
    startActivityForResult(intent, GALLERY_REQUEST_CODE)
}

override fun onActivityResult(requestCode: Int, resultCode: Int, data: Intent?) {
    if (resultCode == Activity.RESULT_OK && requestCode == GALLERY_REQUEST_CODE){
        val selectedUri = data?.data
        val filePathColumn = MediaStore.Images.Media.DATA
        val cursor = context?.contentResolver?.query(selectedUri!!, arrayOf(filePathColumn), null, null, null)
        cursor?.moveToFirst()
        val columnIndex = cursor?.getColumnIndex(filePathColumn)
        val absolutePath = cursor?.getString(columnIndex!!)
        cursor?.close()

        if(absolutePath != null) setImagePathAndPopBack(File(absolutePath))
    }
}

```
This code properly shows the options to pick an image gallery but if I just touch outside to dismiss it, app crashes with above mentioned exception. Sure shot way to reproduce it is to touch the gallery icon and dismiss it a few times without opening any photo gallery. Just to make it clear that majority of the code for onActivityResult is not executed as I am not even picking any image.

build.gradle:

```
implementation "androidx.camera:camera-core:1.0.0-beta02"
implementation "androidx.camera:camera-camera2:1.0.0-beta02"
implementation "androidx.camera:camera-lifecycle:1.0.0-beta02"
implementation "androidx.camera:camera-view:1.0.0-alpha09"
```

Android Studio: 3.6.2 Kotlin: 1.3.61

Thanks in advance for any help!

P.S. : I also noticed that there is a memory leak once I leave the camera fragment. All my attempts to fix that leak went in vein...and I am forced to conclude finally that it's a leak within one of the libraries.

owahltinez, 2020-04-16:
I'm unable to reproduce the issue using the sample without modification

closed 2020-04-16 by owahltinez

knightfork, 2020-04-19:
Not sure if this would reproduce the error, but I am running my code with StrictMode:
        StrictMode.setVmPolicy(
            StrictMode.VmPolicy.Builder()
                .detectLeakedClosableObjects()
                .penaltyLog()
                .penaltyDeath()
                .build()
        )

If I run my code without penaltyDeath() then it works just as smooth as the sample.

knightfork, 2020-05-01:
@owahltinez, did you get a chance to try out the sample with StrictMode? Thanks!