android/camera-samples #204: Could not retrieve native window from surface

procmixin, 2020-04-16:
While running the app, if the phone is locked (power button tapped) and unlock, the app crashes.

<br>Phone: `Pixel 3a`
<br>CameraX version: `1.0.0-beta02`
<br>Commit: 3730442
<br>Stack Trace
> 2020-04-15 19:04:52.864 4300-4300/com.android.example.cameraxbasic E/le.cameraxbasi: Unknown bits set in runtime_flags: 0x8000
2020-04-15 19:04:59.127 4300-4344/com.android.example.cameraxbasic E/Legacy-CameraDevice-JNI: getNativeWindow: Surface had no valid native window.
2020-04-15 19:04:59.127 4300-4344/com.android.example.cameraxbasic E/Legacy-CameraDevice-JNI: LegacyCameraDevice_nativeDetectSurfaceDimens: Could not retrieve native window from surface.
2020-04-15 19:04:59.129 4300-4344/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: CameraX-
    Process: com.android.example.cameraxbasic, PID: 4300
    java.lang.RuntimeException: java.lang.IllegalArgumentException: Surface was abandoned
        at androidx.camera.camera2.internal.Camera2CameraImpl$3.onFailure(Camera2CameraImpl.java:869)
        at androidx.camera.core.impl.utils.futures.Futures$CallbackListener.run(Futures.java:338)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.IllegalArgumentException: Surface was abandoned
        at android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize(SurfaceUtils.java:86)
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:261)
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:146)
        at androidx.camera.camera2.internal.compat.params.OutputConfigurationCompatApi28Impl.<init>(OutputConfigurationCompatApi28Impl.java:34)
        at androidx.camera.camera2.internal.compat.params.OutputConfigurationCompat.<init>(OutputConfigurationCompat.java:51)
        at androidx.camera.camera2.internal.CaptureSession.openCaptureSessionLocked(CaptureSession.java:373)
        at androidx.camera.camera2.internal.CaptureSession.lambda$openCaptureSession$2$CaptureSession(CaptureSession.java:281)
        at androidx.camera.camera2.internal.-$$Lambda$CaptureSession$bwwGuGuBhJx-fgB4Br9Wswwme0U.attachCompleter(Unknown Source:8)
        at androidx.concurrent.futures.CallbackToFutureAdapter.getFuture(CallbackToFutureAdapter.java:102)
        at androidx.camera.camera2.internal.CaptureSession.openCaptureSession(CaptureSession.java:278)
        at androidx.camera.camera2.internal.CaptureSession.lambda$open$0$CaptureSession(CaptureSession.java:242)
        at androidx.camera.camera2.internal.-$$Lambda$CaptureSession$2IbSQd39wMeo2dJgmFG1rvePLoM.apply(Unknown Source:8)
        at androidx.camera.core.impl.utils.futures.ChainingListenableFuture.run(ChainingListenableFuture.java:201)
        at android.os.Handler.handleCallback(Handler.java:883) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:214) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
     Caused by: android.hardware.camera2.legacy.LegacyExceptionUtils$BufferQueueAbandonedException
        at android.hardware.camera2.legacy.LegacyExceptionUtils.throwOnError(LegacyExceptionUtils.java:73)
        at android.hardware.camera2.legacy.LegacyCameraDevice.getSurfaceSize(LegacyCameraDevice.java:631)
        at android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize(SurfaceUtils.java:84)
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:261) 
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:146) 
        at androidx.camera.camera2.internal.compat.params.OutputConfigurationCompatApi28Impl.<init>(OutputConfigurationCompatApi28Impl.java:34) 
        at androidx.camera.camera2.internal.compat.params.OutputConfigurationCompat.<init>(OutputConfigurationCompat.java:51) 
        at androidx.camera.camera2.internal.CaptureSession.openCaptureSessionLocked(CaptureSession.java:373) 
        at androidx.camera.camera2.internal.CaptureSession.lambda$openCaptureSession$2$CaptureSession(CaptureSession.java:281) 
        at androidx.camera.camera2.internal.-$$Lambda$CaptureSession$bwwGuGuBhJx-fgB4Br9Wswwme0U.attachCompleter(Unknown Source:8) 
        at androidx.concurrent.futures.CallbackToFutureAdapter.getFuture(CallbackToFutureAdapter.java:102) 
        at androidx.camera.camera2.internal.CaptureSession.openCaptureSession(CaptureSession.java:278) 
        at androidx.camera.camera2.internal.CaptureSession.lambda$open$0$CaptureSession(CaptureSession.java:242) 
        at androidx.camera.camera2.internal.-$$Lambda$CaptureSession$2IbSQd39wMeo2dJgmFG1rvePLoM.apply(Unknown Source:8) 
        at androidx.camera.core.impl.utils.futures.ChainingListenableFuture.run(ChainingListenableFuture.java:201) 
        at android.os.Handler.handleCallback(Handler.java:883) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:214) 
        at android.os.HandlerThread.run(HandlerThread.java:67)

owahltinez, 2020-04-16:
Looking at the stack, there is no app code at fault here. Can you please open an issue in https://issuetracker.google.com?

procmixin, 2020-04-22:
I have bumped my local copy of this project's camera version to [1.0.0-beta03](https://developer.android.com/jetpack/androidx/releases/camera#camera-camera2-1.0.0-beta03) and camera-view version to 1.0.0-alpha10. The issue seems to have been [resolved](https://developer.android.com/jetpack/androidx/releases/camera#camera-core-1.0.0-beta03) as of this camera release.

closed 2020-04-23 by owahltinez