android/camera-samples #205: GT-I9508V Focusing can only be successful once

androidfei, 2020-04-16:
Running the camerax demo focusing on Samsung GT-I9508V can only succeed once

android：5.0.1
device：Samsung GT-I9508V
camerax version：1.0.0-beta02

owahltinez, 2020-04-16:
Please provide a little more information for us to be able to find the root cause of this issue:
* Did you modify the sample in any way?
* Please provide concrete steps to reproduce this issue

jidekrm, 2020-04-17:
> Please provide a little more information for us to be able to find the root cause of this issue:
> 
>     * Did you modify the sample in any way?
> 
>     * Please provide concrete steps to reproduce this issue

I observed similar problem. I think the issue has to with the last commit(CameraX beta2 #198 ).
I fixed the issue by undoing changes in that commit.

owahltinez, 2020-04-23:
CameraX was updated to beta03, please reopen if you still are seeing an issue and provide more concrete information

closed 2020-04-23 by owahltinez