android/camera-samples #207: YuvToRgbConverter sample incorrect working

AlexanderMatveev, 2020-04-18:
YuvToRgbConverter class generates wrong RGB image:

Creating converter and bitmap:
```kotlin
yuvToRgbConverter = YuvToRgbConverter(applicationContext)
previewBitmap =
            Bitmap.createBitmap(previewSize.width, previewSize.height, Bitmap.Config.RGB_565)
```

Converting ImageProxy image to RGB using example code (https://developer.android.com/training/camerax/analyze):

```kotlin
override fun analyze(imageProxy: ImageProxy, rotationDegrees: Int) {
// ...
yuvToRgbConverter.yuvToRgb(imageProxy.image!!, previewBitmap)
```

Gives result (upper is original TextureView, lower is RGB bitmap result):

[image]

owahltinez, 2020-04-18:
Please see [this file](https://github.com/owahltinez/camerax-tflite/blob/master/app/src/main/java/com/android/example/camerax/tflite/CameraActivity.kt) for sample usage. If you are still seeing issues then, unfortunately, it is likely a device bug in the implementation of the YUV to RGB converter and there is not much we can do about it.

AlexanderMatveev, 2020-04-18:
@owahltinez yes, it's working. Thank you for fast response. I'm closing this issue and going to research example you gave.

closed 2020-04-18 by AlexanderMatveev

AlexanderMatveev, 2020-04-18:
My mistake was expecting `ImageAnalysis.Analyzer` to return `image` with width and height I'm setting with Preview.Builder's `setTargetResolution` and ImageAnalysis.Builder's `setTargetResolution`. I was creating my bitmap with size not same as Analyzer image.