android/camera-samples #209: Crash on locking phone in CameraXBasic Sample

EllingtonKirby, 2020-04-21:
Pixel 4, Android 10, Build Number: QQ2A.200305.003

Steps to reproduce: run the CameraXBasic sample, lock and unlock the screen. After a few cycles of this, the app will crash with the following stack trace. 

`2020-04-21 09:43:05.137 11793-11836/com.android.example.cameraxbasic E/Legacy-CameraDevice-JNI: LegacyCameraDevice_nativeDetectSurfaceDimens: Could not retrieve native window from surface.
2020-04-21 09:43:05.140 11793-11836/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: CameraX-
    Process: com.android.example.cameraxbasic, PID: 11793
    java.lang.RuntimeException: java.lang.IllegalArgumentException: Surface was abandoned
        at androidx.camera.camera2.internal.Camera2CameraImpl$3.onFailure(Camera2CameraImpl.java:869)
        at androidx.camera.core.impl.utils.futures.Futures$CallbackListener.run(Futures.java:338)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.IllegalArgumentException: Surface was abandoned
        at android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize(SurfaceUtils.java:86)
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:261)
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:146)
        at androidx.camera.camera2.internal.compat.params.OutputConfigurationCompatApi28Impl.<init>(OutputConfigurationCompatApi28Impl.java:34)
        at androidx.camera.camera2.internal.compat.params.OutputConfigurationCompat.<init>(OutputConfigurationCompat.java:51)
        at androidx.camera.camera2.internal.CaptureSession.openCaptureSessionLocked(CaptureSession.java:373)
        at androidx.camera.camera2.internal.CaptureSession.lambda$openCaptureSession$2$CaptureSession(CaptureSession.java:281)
        at androidx.camera.camera2.internal.-$$Lambda$CaptureSession$bwwGuGuBhJx-fgB4Br9Wswwme0U.attachCompleter(Unknown Source:8)
        at androidx.concurrent.futures.CallbackToFutureAdapter.getFuture(CallbackToFutureAdapter.java:102)
        at androidx.camera.camera2.internal.CaptureSession.openCaptureSession(CaptureSession.java:278)
        at androidx.camera.camera2.internal.CaptureSession.lambda$open$0$CaptureSession(CaptureSession.java:242)
        at androidx.camera.camera2.internal.-$$Lambda$CaptureSession$2IbSQd39wMeo2dJgmFG1rvePLoM.apply(Unknown Source:8)
        at androidx.camera.core.impl.utils.futures.ChainingListenableFuture.run(ChainingListenableFuture.java:201)
        at android.os.Handler.handleCallback(Handler.java:883) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:214) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
     Caused by: android.hardware.camera2.legacy.LegacyExceptionUtils$BufferQueueAbandonedException
        at android.hardware.camera2.legacy.LegacyExceptionUtils.throwOnError(LegacyExceptionUtils.java:73)
        at android.hardware.camera2.legacy.LegacyCameraDevice.getSurfaceSize(LegacyCameraDevice.java:631)
        at android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize(SurfaceUtils.java:84)
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:261) 
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:146) 
        at androidx.camera.camera2.internal.compat.params.OutputConfigurationCompatApi28Impl.<init>(OutputConfigurationCompatApi28Impl.java:34) 
        at androidx.camera.camera2.internal.compat.params.OutputConfigurationCompat.<init>(OutputConfigurationCompat.java:51) 
        at androidx.camera.camera2.internal.CaptureSession.openCaptureSessionLocked(CaptureSession.java:373) 
        at androidx.camera.camera2.internal.CaptureSession.lambda$openCaptureSession$2$CaptureSession(CaptureSession.java:281) 
        at androidx.camera.camera2.internal.-$$Lambda$CaptureSession$bwwGuGuBhJx-fgB4Br9Wswwme0U.attachCompleter(Unknown Source:8) 
        at androidx.concurrent.futures.CallbackToFutureAdapter.getFuture(CallbackToFutureAdapter.java:102) 
        at androidx.camera.camera2.internal.CaptureSession.openCaptureSession(CaptureSession.java:278) 
        at androidx.camera.camera2.internal.CaptureSession.lambda$open$0$CaptureSession(CaptureSession.java:242) 
        at androidx.camera.camera2.internal.-$$Lambda$CaptureSession$2IbSQd39wMeo2dJgmFG1rvePLoM.apply(Unknown Source:8) 
        at androidx.camera.core.impl.utils.futures.ChainingListenableFuture.run(ChainingListenableFuture.java:201) 
        at android.os.Handler.handleCallback(Handler.java:883) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:214) 
        at android.os.HandlerThread.run(HandlerThread.java:67) `

EllingtonKirby, 2020-04-21:
Looks like the solution is just to upgrade the version of CameraX and camera view
https://issuetracker.google.com/issues/154070419

owahltinez, 2020-04-21:
The open PR is for the core module, so `alpha10` is very old news there. We will upgrade the sample to `beta03` (core) and `alpha10` (view) very soon.

closed 2020-04-23 by owahltinez