android/camera-samples #210: How to manually adjust the exposure compensation value in cameraX

liqingtao, 2020-04-21:
val factory = SurfaceOrientedMeteringPointFactory(1.0f, 1.0f)
val createPoint = factory.createPoint(x, y)
                
val actionAE = FocusMeteringAction.Builder(createPoint, FocusMeteringAction.FLAG_AE)
                    .setAutoCancelDuration(3, TimeUnit.SECONDS)
                    .build()
cameraControl.startFocusAndMetering(actionAE)

Through this method, I can only call API to adjust AE automatically. Is there any API that can set custom exposure compensation value

liqingtao, 2020-04-21:
def camerax_version = "1.0.0-alpha09"
    implementation "androidx.camera:camera-core:$camerax_version"

    // CameraX Camera2 extensions
    implementation "androidx.camera:camera-camera2:$camerax_version"

    // CameraX Lifecycle library
    implementation "androidx.camera:camera-lifecycle:$camerax_version"

    // CameraX View class
    implementation "androidx.camera:camera-view:1.0.0-alpha08"

    // CameraX Extensions library
    // implementation "androidx.camera:camera-extensions:1.0.0-alpha05"
    implementation 'androidx.camera:camera-extensions:1.0.0-alpha09'

This is the reference version number

owahltinez, 2020-04-21:
The open PR is for the core module, so `alpha10` is very old news there. We will upgrade the sample to `beta03` (core) and `alpha10` (view) very soon.

closed 2020-04-21 by owahltinez